MarlinFirmware/Marlin #26893: [BUG] error compiling firmware with stock ender 3 lcd

kyon03, 2024-03-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin/src/lcd/dogm/status_screen_DOGM.cpp: In function 'void _draw_hotend_status(heater_id_t, bool)':
Marlin/src/lcd/dogm/status_screen_DOGM.cpp:328:96: error: 'status_active_extruder_indicator_bmp' was not declared in this scope
           u8g.drawBitmapP(_MAX(0, STATUS_HOTEND_X(heater_id) - 6), STATUS_HEATERS_Y + 3, 1, 5, status_active_extruder_indicator_bmp);
                                                                                                ^
Compiling .pio\build\mks_tinybee\src\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp.o
At global scope:

### Version of Marlin Firmware

bugfix

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-03-21:
No Configs
Nothing to go on

Please attach your configuration files.

ellensp, 2024-03-21:
Managed to reproduce it with this, but no way to know if that i how you triggered it without your config files.

minimal config changes from stock

```Diff
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index 49015aeec5..92005cb7f1 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -68,7 +68,7 @@
 
 // Choose the name from boards.h that matches your setup
 #ifndef MOTHERBOARD
-  #define MOTHERBOARD BOARD_RAMPS_14_EFB
+  #define MOTHERBOARD BOARD_RAMPS_14_EEB
 #endif
 
 /**
@@ -208,13 +208,13 @@
 
 // This defines the number of extruders
 // :[0, 1, 2, 3, 4, 5, 6, 7, 8]
-#define EXTRUDERS 1
+#define EXTRUDERS 2
 
 // Generally expected filament diameter (1.75, 2.85, 3.0, ...). Used for Volumetric, Filament Width Sensor, etc.
 #define DEFAULT_NOMINAL_FILAMENT_DIA 1.75
 
 // For Cyclops or any "multi-extruder" that shares a single nozzle.
-//#define SINGLENOZZLE
+#define SINGLENOZZLE
 
 // Save and restore temperature and fan speed on tool-change.
 // Set standby for the unselected tool with M104/106/109 T...
@@ -3043,7 +3043,7 @@
 //
 // Connect to EXP1 on RAMPS and compatible boards.
 //
-//#define CR10_STOCKDISPLAY
+#define CR10_STOCKDISPLAY
 
 //
 // Ender-2 OEM display, a variant of the MKS_MINI_12864
```

code in question is

```cpp
      #if HAS_MULTI_EXTRUDER && DISABLED(SLIM_LCD_MENUS)
        if (active_extruder == heater_id)
          u8g.drawBitmapP(_MAX(0, STATUS_HOTEND_X(heater_id) - 6), STATUS_HEATERS_Y + 3, 1, 5, status_active_extruder_indicator_bmp);
      #endif
```
HAS_MULTI_EXTRUDER is set due to EXTRUDERS 2

but status_active_extruder_indicator_bmp is not defined due to SINGLENOZZLE

The above code needs to be updated to test for SINGLENOZZLE, as single nozzle should only get a single icon 

Apply this diff
```diff
diff --git a/Marlin/src/lcd/dogm/status_screen_DOGM.cpp b/Marlin/src/lcd/dogm/status_screen_DOGM.cpp
index 24165019b4..cb1df91951 100644
--- a/Marlin/src/lcd/dogm/status_screen_DOGM.cpp
+++ b/Marlin/src/lcd/dogm/status_screen_DOGM.cpp
@@ -323,7 +323,7 @@ FORCE_INLINE void _draw_centered_temp(const celsius_t temp, const uint8_t tx, co
 
       } // PAGE_CONTAINS
 
-      #if HAS_MULTI_EXTRUDER && DISABLED(SLIM_LCD_MENUS)
+      #if HAS_MULTI_EXTRUDER && DISABLED(SLIM_LCD_MENUS) && DISABLED(SINGLENOZZLE)
         if (active_extruder == heater_id)
           u8g.drawBitmapP(_MAX(0, STATUS_HOTEND_X(heater_id) - 6), STATUS_HEATERS_Y + 3, 1, 5, status_active_extruder_indicator_bmp);
       #endif
```

ellensp, 2024-03-21:
Issue was added in https://github.com/MarlinFirmware/Marlin/pull/26152

ellensp, 2024-03-21:
PR coming, after it is confirmed this fixes your issue.

thisiskeithb, 2024-03-21:
> PR coming, after it is confirmed this fixes your issue.

Already created one a few weeks ago: https://github.com/MarlinFirmware/Marlin/pull/26834

thisiskeithb, 2024-03-21:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/26833

closed 2024-03-21 by thisiskeithb

kyon03, 2024-03-22:
sorry guys i dont know how to fix im using bugfix

[config.zip](https://github.com/MarlinFirmware/Marlin/files/14715556/config.zip)
[pins_MKS_TINYBEE.zip](https://github.com/MarlinFirmware/Marlin/files/14715558/pins_MKS_TINYBEE.zip)

ellensp, 2024-03-22:
@kyon03 you either have to wait till https://github.com/MarlinFirmware/Marlin/pull/26834 is committed into bugfix
Or take a look at the file changes or that pull request and change your code the same way  (its a one line change)

kyon03, 2024-03-22:
thanks i did this and it complie, now im having a issue with wifi 
[image]