MarlinFirmware/Marlin #26894: [BUG] G28: Axes move at slow rate after homing at (0,0) with ABL

BigBird0000, 2024-03-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I was playing around with my Marlin Configuration (bugfix, latest build) for my Ender 3 Pro
and had something very strange happen.

When I would issue a G28 (HOME), the hotend and bed would move at normal speeds
getting to X=0 Y=0, but then going towards the center, the bed and the hotend moved
very slowly (I could hear the stepper clicks) [This can be seen in the video Ender3Slow].

If I move the hotend (not using G28) it moves along the X axis at normal speed (see file Ender3Normal).

I was wondering if anyone else had something similar happen, or can point me to a direction
to fix this.

The videos and configuration.h, configuration_adv.h are in this dropbox location:
[](https://www.dropbox.com/scl/fo/c2zwnkwr1m7cie2i3nw60/h?rlkey=1r3irm977gv92gkvogsnvfvyo&dl=0l)

thanks very much!

### Bug Timeline

Started happening this morning

### Expected behavior

I expected a faster speed going towards center of bed, and downward Z movement to get bed location

### Actual behavior

Homed to (0,0) at normal speed. Then going to center of bed, the x and y axis movements were slow, and the Z axes did not lower towards the bed.

### Steps to Reproduce

1. Issue G28

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.1.x

### Printer model

Creality Ender 3 Pro

### Electronics

Stock V4.2.2 board

### LCD/Controller

Creality 12864

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[G28Slow.zip](https://github.com/MarlinFirmware/Marlin/files/14712839/G28Slow.zip)

thinkyhead, 2024-06-22:
This might be patched by now (?) but If not then someone will be sure to look at it ahead of the next release. The issue should be fairly obvious to any investigator, making it a good first issue.

thisiskeithb, 2024-06-22:
> ### Additional information & file uploads
> 
> [G28Slow.zip](https://github.com/MarlinFirmware/Marlin/files/14712839/G28Slow.zip)

You've uploaded an .exe
[image]

Here's a .zip with *just* your configs:

[G28Slow.zip](https://github.com/user-attachments/files/15934692/G28Slow.zip)

---

Some things to check:

- Make sure you really have a 4.2.2 board and not a 4.2.7. You would then use `BOARD_CREALITY_V422` or `BOARD_CREALITY_V427` instead of `BOARD_CREALITY_V4` that you have now.
- With `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` enabled, make sure the BLTouch is plugged into the Z MIN/STOP port and not the dedicated PROBE port.

If all that looks correct, then you can move on to the following:

Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code. You will need to port your config/settings to this version.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as [Pronterface](https://github.com/kliment/Printrun/releases/latest), [Repetier Host](https://www.repetier.com/download-now/), or [OctoPrint](https://octoprint.org/download/).
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

closed 2026-10-08 by thinkyhead