MarlinFirmware/Marlin #26895: BLtouch randomly deploys as an berzerk

Palerm0, 2024-03-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi,
I have an problem with my BLtouch.
I hope someone is able to help me.

Now... I know this is a problem that hass been addresd multiple times.
But no clear solution cam out of it.

I'f dusted of my Ender 5. the very old motherboard was way out of date.
so I decided that it was good to start over. So I orderd the Creality 4.2.7 board

Connected all the cables and the BLtouch cabels according the diagramm of "[Teaching Tech](https://www.youtube.com/watch?v=eF060dBEnfs&t=603s)"'s video.

THe BLtouch is connected to the BL_T connector on the 4.2.7 board. (G(Brown), V(Red), In(Yellow))
The black and white cable is connected to the Z-axis switch port

averything is working fine for 3 months. But slowly the BLtouch pin deployd
during print. But now when the printer is idle just after power on the BLtouch is
going bezerk. Randomly in, and out.  I made an video of it [here](https://github.com/MarlinFirmware/Marlin/assets/56974925/e6975e09-9173-46f8-aa82-93369767aaf1). 
I did not changed any thing in the cabels of in the software

Steps already taken:
Somewhere is fount an possible solution that the megnet in the Pin was weard out.
Remove the pin and re-magnatise it. But this does not work. 
Octoprint disconnected	: this does not work
New cables				: this does not work
New BLtouch				: this does not work
Cable seperate from the rest (possible interferance from other cables)	: this does not work

Does anyonbe had the same problem? How did you solved it?
or does anyone has an clue.
Thanks in advanced

Greeting
Palermo
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/14719600/Archive.zip)

### Bug Timeline

just started after 3 month without problems

### Actual behavior

BLtouch pin randomly deploys and re racks 

### Steps to Reproduce

1. Just by turing on the printer with or without octroprint

### Version of Marlin Firmware

2.1.2.1

### Printer model

Ender 5

### Electronics

Creality 4.2.7

### LCD/Controller

LCD12864

### Other add-ons

BLtouch

### Bed Leveling

ABL 3-point

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-03-22:
not a marlin firmware.  firmware doesn't degrade over 3 moths.
Clearly a faulty probe

closed 2024-03-22 by ellensp

Palerm0, 2024-03-22:
Ok. Thanks.
I understand what you say. I's quite logic.
But that would be 3 BLtouch so far.
The strange part is that the first BLtouch got it over time. The other 2 had it right away from the beginning after installation.

I can order a new one. But I'm feeling that would help me.

InsanityAutomation, 2024-03-22:
> Ok. Thanks. I understand what you say. I's quite logic. But that would be 3 BLtouch so far. The strange part is that the first BLtouch got it over time. The other 2 had it right away from the beginning after installation.
> 
> I can order a new one. But I'm feeling that would help me.

It sounds more like wiring than the probe itself. Sometimes it occurs from EMI noise if the wire is run along stepper or heater lines, anything high frequency output.

Palerm0, 2024-03-22:
I`f already removed the BLtouch cable from the sleeve. So its running seperate from the other cables.
But the cable is joining the other cables when intering the box. I'm gonna see if I can change that. so that the cables are realy separate from the others.
also I'm getting an another BLtouch (Original) from a friend. See if the problem slolves that.

Thanks for poining that out.
I'l let you know

Palerm0, 2024-03-27:
Add another BLtouch.  Same problem. Kicked over the print and the pin is bend.
I'l keep on looking.