MarlinFirmware/Marlin #26896: [BUG] Babystepping the Z-Offset uses the move axis step setting instead of the babystep setting

luizbgomide, 2024-03-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

My setup was working flawlessly one year ago on bugfix-2.1.x (commit 9e42480855), last month I tried to update it to the latest of this branch and after manually migrating all the settings it was working but babystepping the Z Probe Offset was misbehaving.

As can be seem on the video below, when trying to move the Z Offset the display shows (correctly) that 0.01mm should be moved, but the actual move uses the 10mm step for axis move setting (as show in the screen) below.

https://github.com/MarlinFirmware/Marlin/assets/18707781/c06e9f91-b85f-4c3f-926d-b022c8f2ab10

I've tried to change every possible setting related to BABYSTEP and Z_PROBE without any luck. Since I just migrated the settings and the display correctly shows what should have happened, this must be a bug introduced in the last year.

Old settings (at 9e42480855)
[old_settings.zip](https://github.com/MarlinFirmware/Marlin/files/14724199/old_settings.zip)

Current settings
[current_settings.zip](https://github.com/MarlinFirmware/Marlin/files/14724190/current_settings.zip)

### Bug Timeline

It started at some point after commit 9e42480855 from one year ago

### Expected behavior

The hotend moves as indicated by the Z Probe Offset.

### Actual behavior

The hotend moves by the step size for the axis move.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Two Tress Bluer

### Electronics

TMC2209 UART

### LCD/Controller

MKS_ROBIN_TFT35

### Other add-ons

BLTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-03-22:
Confirmed in the simulator

It actually moving 10.01

ellensp, 2024-03-22:
I think im onto something.  

Try this diff   (2 changed lines in Marlin/src/lcd/tft/ui_common.cpp)

```DIFF
diff --git a/Marlin/src/lcd/tft/ui_common.cpp b/Marlin/src/lcd/tft/ui_common.cpp
index 944be77ab5..08fbabdfc6 100644
--- a/Marlin/src/lcd/tft/ui_common.cpp
+++ b/Marlin/src/lcd/tft/ui_common.cpp
@@ -71,7 +71,7 @@ void moveAxis(const AxisEnum axis, const int8_t direction) {
     }
   #endif
 
-  const float diff = motionAxisState.currentStepSize * direction;
+  float diff = motionAxisState.currentStepSize * direction;
 
   #if HAS_BED_PROBE
 
@@ -85,6 +85,7 @@ void moveAxis(const AxisEnum axis, const int8_t direction) {
                       , do_probe ? new_probe_offset : hotend_offset[active_extruder].z - bsDiff
                       , new_probe_offset
                     );
+        diff = bsDiff;
         if (WITHIN(new_offs, PROBE_OFFSET_ZMIN, PROBE_OFFSET_ZMAX)) {
           babystep.add_steps(Z_AXIS, babystep_increment);
           if (do_probe)

```

ellensp, 2024-03-22:
nope its close, but it seem to move twice as far as it should...  

Eg if I set 

  #define BABYSTEP_MILLIMETER_UNITS       // Specify BABYSTEP_MULTIPLICATOR_(XY|Z) in mm instead of micro-steps
  #define BABYSTEP_MULTIPLICATOR_Z  0.1       // (steps or mm) Steps or millimeter distance for each Z babystep

it baby stepping moves in steps of 0.2mm vs expected 0.1

my sim configs if someone want to have a look, im taking a break.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/14726997/Configuration.zip)

ellensp, 2024-03-22:
please note with your current settings you have 410 steps/mm for Z with BABYSTEP_MULTIPLICATOR_Z  4 and BABYSTEP_ZPROBE_OFFSET
So each baby step will move the Z 0.009756098 mm, it will take a few clicks to noticeably move the Z axis

luizbgomide, 2024-03-22:
> please note with your current settings you have 410 steps/mm for Z with BABYSTEP_MULTIPLICATOR_Z 4 and BABYSTEP_ZPROBE_OFFSET So each baby step will move the Z 0.009756098 mm, it will take a few clicks to noticeably move the Z axis

Actually it is 410 for the E0 axis, for Z it is 400.

After your last edit is it working or is it still moving double the distance?

ellensp, 2024-03-22:
correct distance now.  yes your correct its 400, its very late here, going a bit cross eyed.

luizbgomide, 2024-03-22:
It is still misbehaving during print, when I adjust the z-offset it seems to dig the hotend into the bed, and even lose control of the rest of the axis, I've seem it move back during a print move, do some sort of "layer skip", etc.

The current version is unusable as it is for my printer.

SyncroSeb, 2024-04-20:
Has a similar problem.
Printer is a CoreXY (Tronxy X5SA Pro Stock motherboard with  STMF446 Chip)
when adjusting the Z-Offset during the first layer the Z axis moved down not in 0.1 but whole mm steps, sometimes going down 20mm or more printing in the air, then going back to initial layer height. (+- Offset?).

The above mentioned changes to the Marlin/src/lcd/tft/ui_common.cpp (using actual bugfix version) worke out fine. Thx a lot

thisiskeithb, 2024-04-26:
This may have been fixed in https://github.com/MarlinFirmware/Marlin/commit/489ef6e5e1aaf94e1786c50841143c894efe6c2c

Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

closed 2024-12-04 by thinkyhead