MarlinFirmware/Marlin #26899: [BUG] Incorrect Static Assertion Fail during compile due to TRAMMING_POINT_XY

objecttothis, 2024-03-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Auto Build Marlin complains about the third tramming point being unreachable by the probe. In this case, it's actually the first and fourth tramming point's which are unreachable by the probe.

Configuration_adv.h
`#define TRAMMING_POINT_XY { {  30, 30 }, { 205,  30 }, { 205, 205 }, { 30, 205 } }`
Configuration.h
``` 
#define X_BED_SIZE 235
#define Y_BED_SIZE 235
#define X_MIN_POS -5 //Account for Micro Swiss DD kit
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE // The probe can reach up to 269 but should never probe outside of X_BED_SIZE - PROBING_MARGIN
#define Y_MAX_POS Y_BED_SIZE
#define NOZZLE_TO_PROBE_OFFSET { 49.5, 1, -2.11 }
#define PROBING_MARGIN 11.4
```

Modifying the tramming points to `#define TRAMMING_POINT_XY { {  49.5, 30 }, { 205,  30 }, { 205, 205 }, { 49.5, 205 } }` causes the build assertion to pass

### Bug Timeline

New to me but judging by complaints, probably not new

### Expected behavior

I expect to receive an accurate and relevant assertion error.

1. `PROBING_MARGIN` in my case is irrelevant.
2. The configured `NOZZLE_TO_PROBE_OFFSET` is not the default.  Replace "default" with "configured" and replace "NOZZLE_TO_PROBE offset" with the correct `NOZZLE_TO_PROBE_OFFSET`.
3. On assertion fail give an accurate notification of which probe points failed.
4. Optional but better is show the minimum/maximum calculated values for failing assertions.

### Actual behavior

I get the following error 
[image]

```
Marlin\src\gcode\bedlevel\../../feature/tramming.h:59:72: error: static assertion failed: TRAMMING_POINT_XY point 3 is not reachable with the default NOZZLE_TO_PROBE offset and PROBING_MARGIN. 
59 | #define _CHECK_TRAM_POINT(N) static_assert(Probe::build_time::can_reach(tramming_points[N]), "TRAMMING_POINT_XY point " STRINGIFY(N) " is not reachable with the default NOZZLE_TO_PROBE offset and PROBING_MARGIN.");
```

### Steps to Reproduce

1. Build with Auto Build Marlin

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Ender-3

### Electronics

4.2.7, CR-touch, Micro-swiss DD w/all metal hotend

### LCD/Controller

Stock

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/14728241/Configuration.zip)

objecttothis, 2024-03-22:
From #21013 I see that the indices for which tramming point is the problem starts at zero, not one.  I'm a software developer and this wasn't clear to me.  That said, this is still an issue, because it should be reporting assert failures at both index 0 and 3.

thisiskeithb, 2024-03-23:
> ### Additional information & file uploads
> 
> [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/14728241/Configuration.zip)

These are not `bugfix-2.1.x` configs.

[image]

Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

objecttothis, 2024-03-23:
> These are not `bugfix-2.1.x` configs.
>
> [image]
>
> Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if [...]

The code it's compiling is the latest bugfix. The config files were from your list of pre-configured config files for the ender 3. I'll diff with the latest configuration files and report back.

thisiskeithb, 2024-03-23:
> The code it's compiling is the latest bugfix.

That is not possible. `bugfix-2.1.x` will not build with the `2.1.2.2` configs that you uploaded here due to a version mismatch.

thisiskeithb, 2024-03-23:
Using the [`Creality/Ender-3/CrealityV427` config from the Configurations `bugfix-2.1.x` branch](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-3/CrealityV427), it builds fine:

```prolog
Linking .pio/build/STM32F103RE_creality/firmware.elf
Checking size .pio/build/STM32F103RE_creality/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  11.2% (used 7312 bytes from 65536 bytes)
Flash: [===       ]  28.5% (used 149428 bytes from 524288 bytes)
Building .pio/build/STM32F103RE_creality/firmware.bin
rename_target([".pio/build/STM32F103RE_creality/firmware.bin"], [".pio/build/STM32F103RE_creality/firmware.elf"])
========================================================= [SUCCESS] Took 40.42 seconds =========================================================

Environment           Status    Duration
--------------------  --------  ------------
STM32F103RE_creality  SUCCESS   00:00:40.419
========================================================= 1 succeeded in 00:00:40.419 =========================================================
 *  Terminal will be reused by tasks, press any key to close it. 
 ```

objecttothis, 2024-03-23:
I think my confusion had been that while I started with the aim of downloading bugfix-2.1.x, I then followed the procedure outlined for Auto Build Marlin, so it downloaded 2.1.2.2.

I downloaded bugfix-2.1.x then merged changes from my submitted configuration files (see attached) and built again.  The problem is the same

```
Marlin\src\gcode\bedlevel\../../feature/tramming.h:59:72: error: static assertion failed: TRAMMING_POINT_XY point 3 is not reachable with the default NOZZLE_TO_PROBE offset and PROBING_MARGIN. 
   59 | #define _CHECK_TRAM_POINT(N) static_assert(Probe::build_time::can_reach(tramming_points[N]), "TRAMMING_POINT_XY point " STRINGIFY(N) " is not reachable with the default NOZZLE_TO_PROBE offset and PROBING_MARGIN.");
      |                                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
```
[image]
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/14731855/Configuration.zip)

objecttothis, 2024-03-23:
also, something is not right with the tramming wizard.  
- The front-left measurement is taken with the probe position x 54 y 15 not the 49.5 and 30 I have specified.
- The front-right measurement is taken with the probe position x 209 y 15 not the 205 and 30 I have specified.
- The back-right measurement is taken with the probe position x 209 y 192 not the 205 and 205 I have specified.
- The back-left measurement is taken with the probe position x 54 y 192 not the 49.5 and 205 I have specified.

objecttothis, 2024-03-23:
Running Motion > Bed Leveling > Bed Tramming probes in still different points, though much closer to the specified points.  The Z-Homing point is not at BED_X_SIZE /2 , BED_Y_SIZE / 2 like I would expect it.  Like my Y values from bed tramming wizard, it too is 15mm forward. of what it should be (Y 102.5 instead of 117.5).  X is also wrong, but 120 instead of 117.5, so not the 4.5mm it's off in bed tramming wizard.

It may be related but bed leveling is probing at y -1 for the first row. I would need to crank up my probe offsets to get it to get back on the board. Doing that would throw these other tramming algorithms off though.

thisiskeithb, 2024-03-23:
> also, something is not right with the tramming wizard.

You have the following enabled in the configs you uploaded here: `#define TRAMMING_POINT_XY { {  30, 30 }, { 205,  30 }, { 205, 205 }, { 30, 205 } }`, not the values you posted.

> Running Motion > Bed Leveling > Bed Tramming probes in still different points,

Yes, because you have the following in your config: `#define BED_TRAMMING_INSET_LFRB { 49.5, 49.5, 49.5, 49.5 }`

> it's actually the first and fourth tramming point

The echoed tramming point index in the static assert could be improved, but that is already mentioned in https://github.com/MarlinFirmware/Marlin/issues/21013. For your other issues, please visit the links in the message below.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2024-03-23 by thisiskeithb