MarlinFirmware/Marlin #26902: [BUG] Documentation on power loss recovery unclear

phazei, 2024-03-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have power loss recovery enabled.  I'm looking at the settings.  I presume it immediately disables the heaters for everything, but it doesn't say if it does or not.  I just want it to move z-axis a couple mm away from the part and stop.

There is this setting:
//#define POWER_LOSS_ZRAISE       2 // (mm) Z axis raise on resume (on power loss with UPS)

The description says that the Z axis is raised on resume?  That doesn't make sense, does that mean it will lower z on power loss and then raise it when you say to resume after it heats up again?  Will it auto home the x and y, and then presume the z is 2mm away and move back that much when restarted?

//#define POWER_LOSS_PURGE_LEN   20 // (mm) Length of filament to purge on resume
//#define POWER_LOSS_RETRACT_LEN 10 // (mm) Length of filament to retract on fail. Requires backup power.

Those have different defaults.  Wouldn't you want to retract the same amount you purge?  Or if the retract length is 10, it automatically unretracts 10, and then pushes out another 20?  If it pushes out, does it do so right above the part, or right after it homes x and y?

#define POWER_LOSS_MIN_Z_CHANGE 0.05 // (mm) Minimum Z change before saving power-loss data

It says it's in mm, but the value is 0.05?  That's 0.05mm? That seems quite small.  Layer heights are usually at least double that.  Does that mean it'll save if you set your z-hop to 0.05mm?  Is that only for if you have PLR on, but haven't defined BACKUP_POWER_SUPPLY?  Then it's constantly writing where it is to the sdcard?  But if you have defined that then it won't write all the time and only when the power pin is triggered?  Is it ignored if the backup power is defined?

I did google for the terms to see if I could find clarification in forums or in the docs, but I couldn't find anything with better clarification.  Thank you

### Bug Timeline

Years ago

### Expected behavior

I look at the configuration_adv file and understand exactly what to set

### Actual behavior

I look at the configuration_adv file and don't understand exactly what to set

### Steps to Reproduce

Look at the config

### Version of Marlin Firmware

latest 2.1.x bugfix

### Printer model

custom

### Electronics

SKR 1.4

### LCD/Controller

reprap

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

that's all

phazei, 2024-07-01:
Could someone please clarify this please?

phazei, 2024-09-30:
yes

closed 2025-01-14 by github-actions[bot]

thinkyhead, 2025-03-20:
### `POWER_LOSS_ZRAISE`
A Z raise may help prevent the nozzle from sitting on the print and leaving a mess at that spot.

But if you don't have a power-loss detection circuit then _**power-loss will not be detected until you reboot**_, and so the Z axis will not raise in the event of a power loss.

Of course, when the power loss routine knows that it had raised the Z axis on power-loss, then it will lower it back down again when the print resumes.

> Wouldn't you want to retract the same amount you purge?

We have thought these things out (!) and we've received some corrections over the years. Of course when the power loss resume routine runs, we have to re-home, among other things, so of course in that process we do our best to avoid purging all over the print.

### `POWER_LOSS_MIN_Z_CHANGE`

This prevents too much writing to the SD card, especially in vase mode (where you probaby don't want to have PLR enabled anyway). Of course, we only consider the Z axis "raised" when an extrusion (not retraction) occurs at the new height. And, if we already did a PLR save for a given height, we don't save again until we see printing at a greater height.

In general, if the (horrid) SD Card-based power-loss resume is working well for you, then you don't need to adjust any of these settings. If you don't like the way these settings cause PLR to behave in your testing, then you should adjust them.