MarlinFirmware/Marlin #26903: [BUG] KILL, : PRINTER HALTED Please Reset

DrUnKeNTiGeR, 2024-03-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin bugfix 2.1.x with ender 2 pro configs compiles and flashes fine when it boots i get KILL, : PRINTER HALTED Please Reset error we tried looking at terminal but octoprint nor prontor face will connect.. 

### Bug Timeline

New when I flashed firmware

### Expected behavior

To boot into a usable state

### Actual behavior

Error at boot

### Steps to Reproduce

1. turn on printer
2. wait
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/14738681/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/14738682/Configuration_adv.txt)

### Version of Marlin Firmware

bugfix 2.1.x

### Printer model

Ender 2 Pro

### Electronics

None

### LCD/Controller

OEM

### Other add-ons

Nome

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2024-03-25:
I’ll mark this as confirmed since we went through some troubleshooting on Discord. Unfortunately, I don’t have an HC32-based board, but it was clearly halting with current `bugfix-2.1.x`, but does *not* halt with Creality’s firmware for the same board/printer.

Tagging @shadow578 to see if they’ve had any reports with this board/printer or other HC32-based machines on current `bugfix-2.1.x`.

DrUnKeNTiGeR, 2024-03-25:
to make it work comment out #define KILL_PIN EXP1_04_PIN in pins_CREALITY_ENDER2P_V24S4.h

boots like a champ now!!

shadow578, 2024-03-25:
seems like the `KILL_PIN` in `pins_CREALITY_ENDER2P_V24S4.h` is set incorrectly, as both `KILL_PIN` and `X_STEP_PIN` are set to pin `PC1`. 
Since commenting the `KILL_PIN` define makes the printer work, i'm assuming that it's the `KILL_PIN` that is wrong, not 
`X_STEP_PIN`).

---

@DrUnKeNTiGeR could you do the following:

(1) check if LCD display, rotary encoder and beeper work? 
If yes, then only the `EXP1_04` pin is wrong.

(2) if possible confirm the pinout of the EXP1 header on your mainboard.
to do so, use a multimeter in continuity mode and test what pins on the MCU connect to the expansion header. 
if the components in (1) all worked, you'd really only have to check pin 4, but it may be a good idea to use the other pins as practice.

please double-check the following:
- the printer should be turned off
- the lcd ribbon is disconnected from the expansion header while probing
- the orientation of the expansion header is correct (next to pin 5, there should be a notch in the plastic)
- the orientation of the MCU is correct (pin 1 is denoted using a 'dot' on the chip, and the text should be right-side up; see image below)
- the text on the MCU should read `HC32F460KETA` or `HC32F460KCTA`
- the MCU should have 16 pins on either side, for a total of 64 pins.

---

EXP1 pinout:

          ------
    PC6  | 1  2 | PC7
    PA2  | 3  4 | PC1 (?!)
    PB13   5  6 | PB14
    PB15 | 7  8 | PB12
    GND  | 9 10 | 5V
          ------

MCU reference image, pin one is marked by a dot on the chip as well as a triangle on the pcb.

[image]

HC32F460KETA pinout reference:

[image]

thisiskeithb, 2024-03-25:
I think `PC1` was added to the LCD pins incorrectly in https://github.com/MarlinFirmware/Marlin/pull/26730/commits/6257962f7bcccbd6abb9fb23d8a3c3294a9b2baa / https://github.com/MarlinFirmware/Marlin/pull/26730#issuecomment-1911011012

I planned to disable `KILL_PIN` once I’m back at a computer and can submit a PR.

ellensp, 2024-03-25:
According to a hires scan of the board I found online https://github.com/n4bstar/ce2p

the back of the boards says this

[image]

This is flipped left to right form as the image is from the back.

According to this page https://gist.github.com/SteveGotthardt/33bdef27d1430f62b4fee5d723d0cde9
These are the pins (converted for marlin)

```
      ------
PC6  | 1  2 | PC7 
PA2  | 3  4 | PA3 
PB13   5  6 | PB14
 NC    | 7  8 | PB12
GND  | 9 10 | 5V
      ------
```

ellensp, 2024-03-25:
@DrUnKeNTiGeR please confirm the LCD still work  with these new pins and it doesn't HALT

ellensp, 2024-03-26:
PR created, will be added to Marlin code base soon.

closed 2024-03-26 by ellensp