MarlinFirmware/Marlin #26905: [BUG] Backlash measurement during probing should be on first (fast) probe not last (slow) one

Sophist-UK, 2024-03-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Currently if you enable MEASURE_BACKLASH_WHEN_PROBING when you also have MULTIPLE_PROBING >= 2 the backlash measurement is done after the last probe i.e. after a PROBE movement at Z_PROBE_FEEDRATE_SLOW. 

### Bug Timeline

Old issue

### Expected behavior

Backlash is best measured when stopping after a fast movement and it would probably be better done after the first probe which is done at Z_PROBE_FEEDRATE_FAST.

### Actual behavior

Z-Backlash is measured at zero on my machine. I have no idea whether this is correct or not, but it may be due to the measurement being taken at at the slow feedrate. 

### Steps to Reproduce

1. Enable MULTIPLE_PROBING >=2 and MEASURE_BACKLASH_WHEN_PROBING.
2. Do a bed level and see what the backlash is now set to.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Dagoma DiscoEasy200

### Electronics

MKS Base 1.5

### LCD/Controller

REPRAPDISCOUNT_FULL_GRAPHICS

### Bed Leveling

ABL Linear grid

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-03-25:
I disagree with you supposition.

If you moving  fast then stop or change direction, the belts will bounce and move back further than the real backlash
Same if you have those dreadful spring z couples (which the internet says the  Dagoma DiscoEasy200 has) 

This is best done slowly.

Sophist-UK, 2024-03-25:
@ellensp 

My guess is that:

1. Backlash testing is probably best done at the speed used for your z-hops because this is where you want to cancel the backlash.
2. Probe accuracy is best done at a slow speed in order than backlash is zero - hence having a second slower speed for 2nd and subsequent probe measurements at the same spot.

But this is just a guess as someone who has a school / uni background in physics but who is a relative novice to 3D printing.

(And yes, the DE200 does indeed have a spring z-couple, but I have carefully adjusted the backlash nuts on mine so that they are as close to the standard nuts as possible (i.e. within the 60degree limitations of the hex nut fitting into the hex hole) in order to physically limit the range of backlash movements.)

Also, I suspect that Z-backlash (which is what is measured here) is going to be very different for printers that use screws or chains and those that use rubber belts.

I also wonder whether the same measurements could be done on X and Y homing.

bmourit, 2024-03-25:
Maybe there should be introduced a new parameter for backlash measurement that is user adjustable and user can set to the print speed?

Sophist-UK, 2024-06-24:
Please keep this open.

closed 2024-10-07 by github-actions[bot]