MarlinFirmware/Marlin #26907: [BUG] AnycubicKosselLinearPlus-NonPulley_Error-HomingFailedPrinterHaltPleaseReset

AkagiVoid, 2024-03-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Reporting odd behavior bug for G28 auto home in example code for my kossel Linear Plus non pulley version. >w>
[image]

### Bug Timeline

Possibly a New bug, 2024-03-26

### Expected behavior

Auto home Z axis. >w>

### Actual behavior

The rod carriage on all three horizontal bars stops within 5mm from hitting the switch lever, and printer triggers a fail safe for no reason. >w>

### Steps to Reproduce

Step1: Download latest Marlin and install. :P

Step2: Download latest Arduino and install. :3

Step3: Download latest Marlin 2.1.2.2 Example Configurations, extract, open extract folder to "config\examples\delta\Anycubic\Kossel Linear Plus" copy the .h files. >w>

Step4: pasted the copied files directly into "Marlin" folder then open the .ino file with Arduino. >w>

Step5: Calibrate settings to match what I know about my Kossel like the 96 Esteps should be 155 for mine since I'm running a pancake motor, Diagonal Rods are 271.5 (Did change this back to default 267.0 and same homing failure), and set z probe offset to 17.0 (used a caliper and roughly eye balled it). <w<

Step6: Check the compiling codes, then upload it to printer if no errors. >w<

Step7: Do auto bed leveling with probe. :3

Step8: Fine tune Z Offset with paper roughly the size of 0.15mm (kept getting auto home leveling field error). XwX

### Version of Marlin Firmware

2.1.2.2

### Printer model

Kossel Linear Plus

### Electronics

Stock

### LCD/Controller

Stock

### Other add-ons

N/A

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[KosselLinearPlus-NonPulley.zip](https://github.com/MarlinFirmware/Marlin/files/14752930/KosselLinearPlus-NonPulley.zip)

thisiskeithb, 2024-03-26:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as [Pronterface](https://github.com/kliment/Printrun/releases/latest), [Repetier Host](https://www.repetier.com/download-now/), or [OctoPrint](https://octoprint.org/download/).
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

ellensp, 2024-03-26:
homing z moves 1.5 * z height, which your config says is 300mm,  so a z home attempts to move 450mm and expects to hit an endstop or it errors if they are not triggered. 

I suspect your steps/mm is just not correct, since its calculated from these, one or more of these is probably incorrect for your machine.
#define XYZ_FULL_STEPS_PER_ROTATION 200
#define XYZ_MICROSTEPS 16
#define XYZ_BELT_PITCH 2
#define XYZ_PULLEY_TEETH 20

Do a simple test, move all carriage up high and home Z
tell marlin to move down a length say 50mm  (G91 then G1 Z-100) .  measure how far it actually moved

AkagiVoid, 2024-03-26:
@thisiskeithb 
This time, doing the g29 makes the delta go to the front of the printer's bed and take z level on the bed instead of center z level for the center of the bed. >:/
Like it probe goes between the right and left horizontal bars where the LCD is facing and takes a z reading there. ?w?

And redoing the bug fix seem to have fixed the g28 problem but produces a new problem of inconsistent g0 z0 f500 when I try to do offset. XwX
Z offset -17.00 nozzle hits bed belts slip, Z offset 0.00 Nozzle hits bed belts slip, Z Offset +17.00 do not hit bed, Z offset +15.00 do not hit bed, +10 Z offset do not hit bed, Z offset +5.00 Do not hit bed, Z offset 0.00 do not hit bed, Z offset -5.00 hits bed and NO BELT SLIP? @w@ *Confused*

Log 1 after following UR steps. >w>
[KosselLinearPlus_LOG2024-03-27.txt](https://github.com/MarlinFirmware/Marlin/files/14762428/KosselLinearPlus_LOG2024-03-27.txt)

Log 2 after your steps, removed probe, then reboot printer to do z offset check **before** bed leveling:
[KosselLinearPlus-ZOffsetInconsistency_LOG2024-03-27.txt](https://github.com/MarlinFirmware/Marlin/files/14762677/KosselLinearPlus-ZOffsetInconsistency_LOG2024-03-27.txt)

Firmware after applying bugfixs and edit the codes for my printer:
[Mediafire](https://www.mediafire.com/file/qoe8yt0q4wulx4f/KosselLinearPlusMarlin_2024-03-27.zip)

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
@ellensp The output result is 49.69mm after measuring three times. >w>

thisiskeithb, 2024-03-27:
> Log 1 after following UR steps. >w> [KosselLinearPlus_LOG2024-03-27.txt](https://github.com/MarlinFirmware/Marlin/files/14762428/KosselLinearPlus_LOG2024-03-27.txt)
>
> Log 2 after your steps, removed [...]

Both of these logs show that you're still running Marlin `2.1.2.2` which is quite buggy.

Please follow the steps listed in [my previous comment](https://github.com/MarlinFirmware/Marlin/issues/26907#issuecomment-2019441886) *exactly*, which includes downloading the latest `bugfix-2.1.x` to test with the latest code.

AkagiVoid, 2024-03-30:
@thisiskeithb 
Um, using just the bugfix link provided by you, it did dis in log report while compile testing. XwX
[ArduinoIDE_CompileTestLog.txt](https://github.com/MarlinFirmware/Marlin/files/14812540/ArduinoIDE_CompileTestLog.txt)

Foxies-CSTL, 2024-04-03:
@AkagiVoid 
For my Deltas to work properly using the latest Bugfix, I must avoid applying #26297 to the sources, otherwise the calibration (G33) and probe test (M48) will cause measurement errors.

If it helps.

thisiskeithb, 2024-05-08:
> For my Deltas to work properly using the latest Bugfix, I must avoid applying #26297 to the sources, otherwise the calibration (G33) and probe test (M48) will cause measurement errors.

@Foxies-CSTL: Can you apply just the changes from initial commit in that PR? https://github.com/MarlinFirmware/Marlin/pull/26297/commits/a72764ee3b189d4a7fa94ca75547f830572ac5bf

<details><summary>diff:</summary>
<p>

```diff
diff --git a/Marlin/src/module/delta.cpp b/Marlin/src/module/delta.cpp
index cfa134860b86..b624396d5217 100644
--- a/Marlin/src/module/delta.cpp
+++ b/Marlin/src/module/delta.cpp
@@ -242,7 +242,7 @@ void home_delta() {
   #endif
 
   // Move all carriages together linearly until an endstop is hit.
-  current_position.z = DIFF_TERN(HAS_BED_PROBE, delta_height + 10, probe.offset.z);
+  current_position.z = TERN(HAS_BED_PROBE, DIFF_TERN(USE_PROBE_FOR_Z_HOMING,  delta_height + 10, probe.offset.z), delta_height + 10);
   line_to_current_position(homing_feedrate(Z_AXIS));
   planner.synchronize();
   TERN_(HAS_DELTA_SENSORLESS_PROBING, endstops.report_states());
diff --git a/Marlin/src/module/motion.cpp b/Marlin/src/module/motion.cpp
index 272b16c9ac0f..e96c2ca4d5c0 100644
--- a/Marlin/src/module/motion.cpp
+++ b/Marlin/src/module/motion.cpp
@@ -892,7 +892,7 @@ void restore_feedrate_and_scaling() {
     #elif ENABLED(DELTA)
 
       soft_endstop.min[axis] = base_min_pos(axis);
-      soft_endstop.max[axis] = (axis == Z_AXIS) ? DIFF_TERN(HAS_BED_PROBE, delta_height, probe.offset.z) : base_max_pos(axis);
+      soft_endstop.max[axis] = (axis == Z_AXIS) ? TERN(HAS_BED_PROBE, DIFF_TERN(USE_PROBE_FOR_Z_HOMING, delta_height, probe.offset.z) , delta_height) : base_home_pos(axis);
 
       switch (axis) {
         case X_AXIS:
@@ -2468,7 +2468,7 @@ void set_axis_is_at_home(const AxisEnum axis) {
   #if ANY(MORGAN_SCARA, AXEL_TPARA)
     scara_set_axis_is_at_home(axis);
   #elif ENABLED(DELTA)
-    current_position[axis] = (axis == Z_AXIS) ? DIFF_TERN(HAS_BED_PROBE, delta_height, probe.offset.z) : base_home_pos(axis);
+    current_position[axis] = (axis == Z_AXIS) ? TERN(HAS_BED_PROBE, DIFF_TERN(USE_PROBE_FOR_Z_HOMING, delta_height, probe.offset.z) , delta_height) : base_home_pos(axis);
   #else
     current_position[axis] = SUM_TERN(HAS_HOME_OFFSET, base_home_pos(axis), home_offset[axis]);
   #endif
```

</p>
</details>

If that doesn't work, I'll submit a PR to fully revert #26297.

Foxies-CSTL, 2024-05-11:
> @Foxies-CSTL: Can you apply just the changes from initial commit in that PR? [a72764e](https://github.com/MarlinFirmware/Marlin/commit/a72764ee3b189d4a7fa94ca75547f830572ac5bf)
> diff:
> 
> If that doesn't work, I'll submit a PR to fully revert #26297.

Test on my Delta QQSP (BugFix 2.1.3 - Bump 2024/04/29):

[image]

I don't have the same behavior, but the calibration is still not complete. so this doesn't solve the problem.

https://github.com/MarlinFirmware/Marlin/assets/24864147/65bf11ab-9184-43ef-8e95-9507d1a2eda5

Sorry about the quality.
and the function file activated in the firmware.
[20240511-Config-BugG29.txt](https://github.com/MarlinFirmware/Marlin/files/15283128/20240511-Config-BugG29.txt)

thisiskeithb, 2024-05-12:
PR to revert #26297 has been submitted:
- https://github.com/MarlinFirmware/Marlin/pull/27088

closed 2024-05-17 by thinkyhead