MarlinFirmware/Marlin #26913: [BUG] SD Card is detected but won't start printing or heating

Hydra99o, 2024-03-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I've flashed the latest marlin bugfix 2.1.x to my ANYCUBIC CHIRON, when i try to start a print via SD card using the lcd menu i can see the files in the SD card files menu , after selecting a file it takes me to the status screen as usual but it doesn't do anything, it shows the file name but wont start printing
Note: It prints fine with USB and i can also start a print from SD card using Pronterface the problem only occurs when trying to print using the stock Chiron LCD 
Also in Pronterface log it says SD Card ok "echo:SD card ok"
this bug doesn't exist in previous versions of marlin 

### Bug Timeline

NEW BUG

### Expected behavior

I expect it to start heating and printing after selecting a file form the menu 

### Actual behavior

it wont start printing

### Version of Marlin Firmware

Bugfix 2.1.x

### Printer model

Anycubic Chiron

### Electronics

stock Trigorilla_14

### LCD/Controller

ANYCUBIC LCD CHIRON

### Other add-ons

no add-ons except a direct drive extruder 

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/14760774/Configuration_adv.zip)

mechano, 2024-03-31:
Yes same as mine, with Kingroon machines with Kingroon 1.3 board. It's an MKS Robin Nano 1.2 clone.

closed 2024-09-21 by github-actions[bot]

SMH17, 2024-10-07:
This bug is still present.

carloslockward, 2024-10-08:
Im experiencing this bug aswell. Kingroon KP3S with Marlin compiled from the tagged `2.1.2.4` release. Please re-open.

SMH17, 2024-10-09:
If you use FSCM and an STM32 based motherboard the issue may be caused by [this ](https://github.com/MarlinFirmware/Marlin/issues/27460).