MarlinFirmware/Marlin #26916: [BUG] Wrong pin mapping for Filament Runout Sensor on SKR 1.4 Turbo and BTT SFS V2.0

RoscoeTheDog, 2024-03-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have a SKR 1.4 Turbo board with TMC 2209 drivers. To get the SFS V2.0 motion tracking and filament runout sensor working with the latest Marlin (2.1.2) there were several things required:

1) TMC2209 drivers or above need the 'diag' pin disconnected for E0 stepper drivers. This is a requirement for all other endstops to work as well, and has been documented in the BTT github repo for the BTT TMC2209 V3 stepper drivers. This is not a bug, but worth mentioning for readers having problems.

2) The BTT documentation states to plug the motion sensor pins to E0DET and the switch sensor to to E1DET. This is WRONG. They must be swapped. This took me a very long time to figure out and was the main reason for troubleshooting issues. The switch sensor needs to be plugged into E0DET and motion sensor to E1DET, or you can re-map them digitally like I did by explicitly declaring the pin outputs for FIL_RUNOUT_PIN and FIL_RUNOUT_PIN2 respectively.

3) I had to invert the endstop logic for the filament runout sensor. Instead of LOW I had to have it HIGH.

Here is my configuration.h snippit for enabling the sensor. I hope it saves other people more time with other boards. Keep in mind I remapped the E0DET and E1DET pins instead of physically swapping them on the board.

I am not sure if this is a bug with Marlin 2.1.x or if it is a bug with poor documentation on behalf of BTT's part. Maybe a moderator or developer can clarify. Thanks 🙏 

```
#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define FIL_RUNOUT_ENABLED_DEFAULT true // Enable the sensor on startup. Override with M412 followed by M500.
  #define NUM_RUNOUT_SENSORS   1          // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.

  #define FIL_RUNOUT_STATE     HIGH        // Pin state indicating that filament is NOT present.
  #define FIL_RUNOUT_PULLUP               // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN           // Use internal pulldown for filament runout pins.
  //#define WATCH_ALL_RUNOUT_SENSORS      // Execute runout script on any triggering sensor, not only for the active extruder.
                                          // This is automatically enabled for MIXING_EXTRUDERs.

  // Override individually if the runout sensors vary
  //#define FIL_RUNOUT1_STATE LOW
  //#define FIL_RUNOUT1_PULLUP
  //#define FIL_RUNOUT1_PULLDOWN

  //#define FIL_RUNOUT2_STATE LOW
  //#define FIL_RUNOUT2_PULLUP
  //#define FIL_RUNOUT2_PULLDOWN

  //#define FIL_RUNOUT3_STATE LOW
  //#define FIL_RUNOUT3_PULLUP
  //#define FIL_RUNOUT3_PULLDOWN

  //#define FIL_RUNOUT4_STATE LOW
  //#define FIL_RUNOUT4_PULLUP
  //#define FIL_RUNOUT4_PULLDOWN

  //#define FIL_RUNOUT5_STATE LOW
  //#define FIL_RUNOUT5_PULLUP
  //#define FIL_RUNOUT5_PULLDOWN

  //#define FIL_RUNOUT6_STATE LOW
  //#define FIL_RUNOUT6_PULLUP
  //#define FIL_RUNOUT6_PULLDOWN

  //#define FIL_RUNOUT7_STATE LOW
  //#define FIL_RUNOUT7_PULLUP
  //#define FIL_RUNOUT7_PULLDOWN

  //#define FIL_RUNOUT8_STATE LOW
  //#define FIL_RUNOUT8_PULLUP
  //#define FIL_RUNOUT8_PULLDOWN

  // Commands to execute on filament runout.
  // With multiple runout sensors use the %c placeholder for the current tool in commands (e.g., "M600 T%c")
  // NOTE: After 'M412 H1' the host handles filament runout and this script does not apply.
  #define FILAMENT_RUNOUT_SCRIPT "M600"

  // After a runout is detected, continue printing this length of filament
  // before executing the runout script. Useful for a sensor at the end of
  // a feed tube. Requires 4 bytes SRAM per sensor, plus 4 bytes overhead.
  #define FILAMENT_RUNOUT_DISTANCE_MM 10

  #ifdef FILAMENT_RUNOUT_DISTANCE_MM
    // Enable this option to use an encoder disc that toggles the runout pin
    // as the filament moves. (Be sure to set FILAMENT_RUNOUT_DISTANCE_MM
    // large enough to avoid false positives.)
    #define FIL_RUNOUT_PIN P1_25
    #define FIL_RUNOUT_PIN2 P1_26
    #define FILAMENT_MOTION_SENSOR
    #define FILAMENT_SWITCH_AND_MOTION
    
  #endif
#endif
```

### Expected behavior

Default configuration to work as indicated by BTT.

### Actual behavior

Constantly reading OPEN. This would cause a print to stop as soon as the specified MM length elapsed.

### Version of Marlin Firmware

2.1.2.2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/14784985/Marlin.zip)

thisiskeithb, 2024-03-28:
The pins you use does not matter as long as they are configured correctly, but `FILAMENT_SWITCH_AND_MOTION` is not supported in 2.1.2.2, so you need to use the latest `bugfix-2.1.x`.

BTT's docs are also usually behind what is supported here in mainline / specific to their fork, which is outdated.

---
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2024-03-28 by thisiskeithb

RoscoeTheDog, 2024-03-28:
> The pins you use does not matter as long as they are configured correctly, but `FILAMENT_SWITCH_AND_MOTION` is not supported in 2.1.2.2, so you need to use the latest `bugfix-2.1.x`.
>
> BTT's docs are [...]

That is really strange because I got `FILAMENT_SWITCH_AND_MOTION` working with the specified configuration and not using bugfix repo. I tested the runout trigger and the motion jamming independently with a small piece of filament, and both features have worked without issues. As mentioned in the OP, I _did_ have to swap the pin outputs and invert the trigger logic, but I did get it working on the main repo.

thisiskeithb, 2024-03-28:
> That is really strange because I got FILAMENT_SWITCH_AND_MOTION working with the specified configuration and not using bugfix repo

Unless you completely backported the feature and code changes to 2.1.2.2, the only reason it “works” is because of your workarounds.

Grab the latest `bugfix-2.1.x` and configuring this sensor will be a lot easier.

Please use the links in my previous comment for further troubleshooting.

skorokithakis, 2024-04-05:
I haven't managed to get this senros to work on my MKS Robin Nano 1.2, with Marlin 2.1. This is my configuration:

```
#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define FIL_RUNOUT_ENABLED_DEFAULT true // Enable the sensor on startup. Override with M412 followed by M500.
  #define NUM_RUNOUT_SENSORS   1          // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.

  #define FIL_RUNOUT_STATE     LOW        // Pin state indicating that filament is NOT present.
  #define FIL_RUNOUT_PULLUP               // Use internal pullup for filament runout pins.
  #define FILAMENT_RUNOUT_SCRIPT "M600"
  #define FILAMENT_RUNOUT_DISTANCE_MM 7

  #ifdef FILAMENT_RUNOUT_DISTANCE_MM
    #define FILAMENT_MOTION_SENSOR
  #endif
#endif
```

Does anyone know why this might not be working? The pins are correct, the switch sensor itself works fine if `FILAMENT_MOTION_SENSOR` is commented out, but the motion sensor isn't working even though I can see the LED blinking as the filament goes through the sensor.