MarlinFirmware/Marlin #26920: Marlin TFT Screen Glitch

nichols89ben, 2024-03-29:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Started getting a screen glitch at random moment, but happens every print at some point. Ive searched forums but havent found a solution that works. I also tried the section that was in troubleshooting 

#define ST7920_DELAY_1 DELAY_NS(200) // After CLK LOW
#define ST7920_DELAY_2 DELAY_NS(400) // After DAT
#define ST7920_DELAY_3 DELAY_NS(200) // After CLK HIGH

[image]
[image]

### Version of Marlin Firmware

bugfix-2.1.x 2024-03-25

### Printer model

SV04 Modified

### Electronics

BTT Octopus V1.1

### LCD/Controller

TFT70

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/14807544/Archive.zip)

thisiskeithb, 2024-03-29:
More tuning of the `ST7920_DELAY_*` values are needed and/or you should shield your LCD cables / move them away from motor / power wiring.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2024-03-29 by thisiskeithb

nichols89ben, 2024-03-29:
> More tuning of the `ST7920_DELAY_*` values are needed and/or you should shield your LCD cables / move them away from motor / power wiring.
>
> This Issue Queue is for Marlin bug reports and [...]

Thanks, Do you suggest a value or a particular shielding to wrap the connectors in? I know you guys always ask to go to the forums, but out of all the questions ive asked there Ive yet to ever receive a response. Not an excuse, just wanted to clarify.