MarlinFirmware/Marlin #26921: After Power loss printer don't offer resume printing, with USB adapter FYSETC 3421 on MAX3421E chip.

Deema35, 2024-03-30:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I use USB adapter FYSETC 3421 on MAX3421E chip. After Power loss printer don't offer resume printing. I overcame this by adding line in cardreader.cpp. And now if flash drive has PLR file, printer offer resume printing.
[image]

### Bug Timeline

New

### Expected behavior

After Power loss printer offer resume printing.

### Actual behavior

After Power loss printer don't offer resume printing.

### Steps to Reproduce

Add to printer FYSETC 3421 USB adapter. And during printing power off, and then power on.

### Version of Marlin Firmware

bugfix-2.1.x latest

### Printer model

My own

### Electronics

RUMBA 32

### LCD/Controller

LCD 12864

### Other add-ons

FYSETC 3421 on MAX3421E chip

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/14811331/Config.zip)

thinkyhead, 2024-12-11:
If you use any type of media reader that lacks an SD Detect circuit or behaves differently from the typical SD Card reader, it might help to enable this option:
```cpp
  // Enable if SD detect is rendered useless (e.g., by using an SD extender)
  //#define NO_SD_DETECT
```

Deema35, 2024-12-12:
Not. I have manage out how overcome this bug, I had described it In first post. I created this thread for people who had same issue.

closed 2025-03-27 by github-actions[bot]