MarlinFirmware/Marlin #26923: [BUG] marlin hang on 100%

FlyAAM, 2024-03-30:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

hi, I have Robin nano v3.1 and it works fine with Marlin 2.0.9, but I need input shaping. I created Marlin 2.1.2.2 for my printer and can't update it. The printer starts updating, the status bar appears and shows 100%. and nothing else changes. Green bar and 100%. what can I do to solve this problem?

### Expected behavior

expected the printer to be fully loaded and working

### Actual behavior

marlin 2.1.2.2 hang on 100%

### Steps to Reproduce

use board mks robin nano v3.1 (s407vet6), screen mks TS35 v2 and upload the firmware marlin 2.1.2.2

### Version of Marlin Firmware

2.1.2.2

### Printer model

custom

### Electronics

mks robin nano v3.1

### LCD/Controller

mks TS35 v2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2024-03-30:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/26811

thisiskeithb, 2024-03-30:
Use `bugfix-2.1.x` since it's been fixed there.

closed 2024-03-30 by thisiskeithb

FlyAAM, 2024-03-30:
I downloaded bugfix 2.1 yesterday.. it also doesn’t work. Maybe it's a download error, I'll double check

thisiskeithb, 2024-03-30:
`2.1.2.2` is not `bugfix-2.1.x`.

You'll need to use [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip).

FlyAAM, 2024-03-30:
Sorry, I was in a hurry. I downloaded and used bugfix 2.1.x yesterday but did not achieve any results

thisiskeithb, 2024-03-30:
You didn't provide your configs, but I do not have issues with the Robin Nano V3.1 & TS35 on current `bugfix-2.1.x`.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).