MarlinFirmware/Marlin #26924: Mendel90: Problem Firmware compiling

stevie72, 2024-03-30:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello Marlin Team
I´m using an old outdated MENDEL90 with Magnetic Z-Probe for Bedlevelig, Reprap Full Grafic Controller and RAMPS1.4
I dont get the Firmware compiled. Please look the added Screenshot
BTW .. the good old MENDEL90 is missing in the Configs Examples ..

[image]

### Bug Timeline

After enabling the Auto Bed Leveling Feature

### Version of Marlin Firmware

Marlin 2.1.1.1

### Printer model

MENDEL90

### LCD/Controller

Reprap Full Grafics Controller 

### Other add-ons

RAMPS1.4

### Bed Leveling

ABL 3-point

### Your Slicer

Other (explain below)

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Archiv.zip](https://github.com/MarlinFirmware/Marlin/files/14813334/Archiv.zip)

stevie72, 2024-03-30:
I´ve found the bug ,,
In the status_screen_DOGM.cpp line 506 is marked with  // static u8g_uint_t progress_bar_solid_width = 0;
Remove the // and it works 

Cheers Stefan

thisiskeithb, 2024-03-30:
> Version of Marlin Firmware
> Marlin 2.1.1.1

You’re not using the latest code.

Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

thisiskeithb, 2024-03-30:
This has already been fixed in `bugfix-2.1.x`:

https://github.com/MarlinFirmware/Marlin/blob/17594298e194b14ee62d32d5574284b6b2c9eb5a/Marlin/src/lcd/dogm/status_screen_DOGM.cpp#L554-L556

closed 2024-03-30 by thisiskeithb