MarlinFirmware/Marlin #26927: SD read broken on MKS Robin Nano 1.2 (and clones) after 14 Jan 2024

mechano, 2024-03-31:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I've compiled successfully the 14 Jan 2024 version and is fully working. 
Now I'm still able to compile the last bugfix, but on Kingroon KP3S (Pro and S1), KP5L machines and other Robin Nano 1.2 machines the onboard microSD reader doesn't work anymore. 
I can see file list, I choose one and the print doesn't start. 
When I go back in the file list the filenames are 8.3 characters and no folders are read. 

It prints from Octoprint. 

### Bug Timeline

New one

### Expected behavior

Read files and print them.

### Actual behavior

Read file but doesn't print. Going back to file list the directory structure is broken.

### Version of Marlin Firmware

2.1.2-bugfix after 14 Jan 2024

### Printer model

Kingroon KP3S Pro and KP5L

### Electronics

Stock Robin Nano 1.2 clone boards.

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/14817819/Configuration.zip)

Hero3dPrints, 2024-04-02:
Same problem here. Read fails under random conditions. Reverting to mks firmware has no issue.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/14842366/Marlin.zip)

avizienis, 2024-04-10:
Same here with the current 2.1.x bugfix version. Haven't tried the stable ones

FabianMPunkt, 2024-05-30:
Same here on Elegoo Neptune 2s with ZNP robon nano V1.3
Issue appears on release 2.1.2.2, realease 2.1.2.3, and current bugfix branch.
In my case the printer sometimes does random travels or reboots in the middle of the print. Only when printing from SDCards. Tried multiple SDCards, same issue every time. Works fine when printing via OctoPrint.
IT also works just fine when i enable TFT_LVGL_UI

AR1972, 2024-06-16:
same issue here with the latest commit on a KPS pro s1

andrademf, 2024-07-26:
Same here on MKS Robin Nano V3.1 with bugfix-2.1.x at ad60cdd03067877bfa453a0b18b2dd044802b55e.

I'm no longer able to print from the SD, Marling crashes right after homing

konradmb, 2024-08-31:
Here's the fix: #27385

closed 2024-09-05 by thinkyhead