MarlinFirmware/Marlin #26930: [BUG] Change filament raises Z without knowing position

Sophist-UK, 2024-04-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Using `Unload filament` or `Change filament` (from the menu), if the head is near the bed it starts by raising the head (because it will do a purge and needs to ensure enough height that filament falls away from the nozzle rather than snaking around it).

However if the Z-position is unknown (because a home has not yet been done) and the head is already at (or close to) its maximum, then the head is raised anyway and grinding occurs.

### Bug Timeline

No idea

### Expected behavior

I expect Marlin to do a home before doing an unload filament or change filament if the Z-position is unknown.

### Actual behavior

Home is not done and if head is already at Z-max grinding occurs.

### Steps to Reproduce

1. Raise head to maximum height.
2. Turn off machine.
3. Turn on machine.
4. Do Unload Filament from the menus.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Dagoma DiscoEasy200

### Electronics

MKS Base 1.5 AVR

### LCD/Controller

REPRAPDISCOUNT_FULL_GRAPHICS

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin-DE200-2.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/14837034/Marlin-DE200-2.1.x.zip)

ellensp, 2024-04-02:
Enable 
//#define NO_MOTION_BEFORE_HOMING // Inhibit movement until all axes have been homed. Also enable HOME_AFTER_DEACTIVATE for extra safety.

thisiskeithb, 2024-04-02:
> Using Unload filament or Change filament (from the menu), if the head is near the bed it starts by raising the head (because it will do a purge and needs to ensure enough height that filament falls away from the nozzle rather than snaking around it).

Set `NOZZLE_PARK_Z_RAISE_MIN` to `0`. You current have it set to `5`:

```c
#define NOZZLE_PARK_Z_RAISE_MIN   5   // (mm) Always raise Z by at least this distance
```

> ### Expected behavior
> 
> I expect Marlin to do a home before doing an unload filament or change filament if the Z-position is unknown.

Then you need to enable `HOME_BEFORE_FILAMENT_CHANGE`

closed 2024-04-02 by thisiskeithb

Sophist-UK, 2024-04-02:
Nope - this is not the fix. I already allow 5mm above the Z-max for park and home etc. initial raises - the unload/change filament raise is significantly more than 5mm.

(And I need to have this so that it actually raises by 5mm for a Park.)

Sophist-UK, 2024-04-02:
@thisiskeithb @ellensp 

I have now had the chance to look at the code and `HOME_BEFORE_FILAMENT_CHANGE` is exactly what is needed - it does a home ONLY if axes are NOT trusted.

The existing comment for this definition in Configuration_adv.h does read "If needed, home before parking for filament change", so strictly speaking it does say that it will only home if needed, it just doesn't make it clear that it is needed only if it doesn't know the head position. 

Anyway, I have now enabled this setting and I am happy for this issue to remain closed.

Thanks to @ellensp for pointing me to the exact setting which does indeed do exactly what I was asking for.

Sophist-UK, 2024-04-02:
P.S. Although I have previously reviewed all settings, with hindsight I feel I should have double checked them before creating this issue. I am not sure whether I would have definitely believed that this setting would fix the issue, but I think I would at least have investigated the code or tried the setting. So apologies for having made a bug report for what is a configuration error.