MarlinFirmware/Marlin #26934: [BUG] Compiling errors when defining additional axes as rotational

mh-alinaghipour, 2024-04-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

While compiling a configuration containing two additional rotational axes, I faced the errors below:

```
Compilation error: 'struct Planner::_populate_block(block_t*, const abce_long_t&, const xyze_pos_t&, feedRate_t, uint8_t, const PlannerHints&)::DistanceMM' has no member named 'k'
'struct Planner::_populate_block(block_t*, const abce_long_t&, const xyze_pos_t&, feedRate_t, uint8_t, const PlannerHints&)::DistanceMM' has no member named 'u'
'struct Planner::_populate_block(block_t*, const abce_long_t&, const xyze_pos_t&, feedRate_t, uint8_t, const PlannerHints&)::DistanceMM' has no member named 'v'
'struct Planner::_populate_block(block_t*, const abce_long_t&, const xyze_pos_t&, feedRate_t, uint8_t, const PlannerHints&)::DistanceMM' has no member named 'w'
```
I was able to fix the earlier errors by adding this:
`#define HAS_ROTATIONAL_AXES "n"`
to `configuration.h` in which `"n"` represents the number of rotational axes you have defined in "stepper drivers" section.

However, compiling the above, caused new set of errors:
```
line: 2864  error: missing binary operator before token "("
   #if __has_cpp_attribute(fallthrough)
line: 3859  error: missing binary operator before token "("
   #if __has_cpp_attribute(fallthrough)
```
The errors are caused by pragma GCC diagnostics codes in `"\Marlin-2.1.2.2\Marlin\src\module\temperature.cpp"`. To suppress pragma GCC diagnostics errors I did the changes below to `temperature.cpp`:
```
++#pragma GCC diagnostic push
++#pragma GCC diagnostic ignored "-Wimplicit-fallthrough"
#if HAS_THERMAL_PROTECTION

  --#pragma GCC diagnostic push
  --#if __has_cpp_attribute(fallthrough)
    --#pragma GCC diagnostic ignored "-Wimplicit-fallthrough"
  --#endif

  Temperature::tr_state_machine_t Temperature::tr_state_machine[NR_HEATER_RUNAWAY]; // = { { TRInactive, 0 } };

  /**
   * @brief Thermal Runaway state machine for a single heater
   * @param current          current measured temperature
   * @param target           current target temperature
   * @param heater_id        extruder index
   * @param period_seconds   missed temperature allowed time
   * @param hysteresis_degc  allowed distance from target
   *
   * TODO: Embed the last 3 parameters during init, if not less optimal
   */
  void Temperature::tr_state_machine_t::run(const_celsius_float_t current, const_celsius_float_t target, const heater_id_t heater_id, const uint16_t period_seconds, const celsius_float_t hysteresis_degc) {

    #if HEATER_IDLE_HANDLER
      // Convert the given heater_id_t to an idle array index
      const IdleIndex idle_index = idle_index_for_id(heater_id);
    #endif

    /**
      SERIAL_ECHO_START();
      SERIAL_ECHOPGM("Thermal Runaway Running. Heater ID: ");
      switch (heater_id) {
        case H_BED:     SERIAL_ECHOPGM("bed"); break;
        case H_CHAMBER: SERIAL_ECHOPGM("chamber"); break;
        default:        SERIAL_ECHO(heater_id);
      }
      SERIAL_ECHOLNPGM(
        " ; sizeof(running_temp):", sizeof(running_temp),
        " ;  State:", state, " ;  Timer:", timer, " ;  Temperature:", current, " ;  Target Temp:", target
        #if HEATER_IDLE_HANDLER
          , " ;  Idle Timeout:", heater_idle[idle_index].timed_out
        #endif
      );
    */

    #if ENABLED(THERMAL_PROTECTION_VARIANCE_MONITOR)
      if (state == TRMalfunction) { // temperature invariance may continue, regardless of heater state
        variance += ABS(current - last_temp); // no need for detection window now, a single change in variance is enough
        last_temp = current;
        if (!NEAR_ZERO(variance)) {
          variance_timer = millis() + SEC_TO_MS(period_seconds);
          variance = 0.0;
          state = TRStable; // resume from where we detected the problem
        }
      }
    #endif

    if (TERN1(THERMAL_PROTECTION_VARIANCE_MONITOR, state != TRMalfunction)) {
      // If the heater idle timeout expires, restart
      if (TERN0(HEATER_IDLE_HANDLER, heater_idle[idle_index].timed_out)) {
        state = TRInactive;
        running_temp = 0;
        TERN_(THERMAL_PROTECTION_VARIANCE_MONITOR, variance_timer = 0);
      }
      else if (running_temp != target) { // If the target temperature changes, restart
        running_temp = target;
        state = target > 0 ? TRFirstHeating : TRInactive;
        TERN_(THERMAL_PROTECTION_VARIANCE_MONITOR, variance_timer = 0);
      }
    }

    switch (state) {
      // Inactive state waits for a target temperature to be set
      case TRInactive: break;

      // When first heating, wait for the temperature to be reached then go to Stable state
      case TRFirstHeating:
        if (current < running_temp) break;
        state = TRStable;

      // While the temperature is stable watch for a bad temperature
      case TRStable: {

        const celsius_float_t rdiff = running_temp - current;

        #if ENABLED(ADAPTIVE_FAN_SLOWING)
          if (adaptive_fan_slowing && heater_id >= 0) {
            const int_fast8_t fan_index = _MIN(heater_id, FAN_COUNT - 1);
            uint8_t scale;
            if (fan_speed[fan_index] == 0 || rdiff <= hysteresis_degc * 0.25f)
              scale = 128;
            else if (rdiff <= hysteresis_degc * 0.3335f)
              scale = 96;
            else if (rdiff <= hysteresis_degc * 0.5f)
              scale = 64;
            else if (rdiff <= hysteresis_degc * 0.8f)
              scale = 32;
            else
              scale = 0;

            fan_speed_scaler[fan_index] = scale;
          }
        #endif // ADAPTIVE_FAN_SLOWING

        const millis_t now = millis();

        #if ENABLED(THERMAL_PROTECTION_VARIANCE_MONITOR)
          if (PENDING(now, variance_timer)) {
            variance += ABS(current - last_temp);
            last_temp = current;
          }
          else {
            if (NEAR_ZERO(variance) && variance_timer) { // valid variance monitoring window
              state = TRMalfunction;
              break;
            }
            variance_timer = now + SEC_TO_MS(period_seconds);
            variance = 0.0;
            last_temp = current;
          }
        #endif

        if (rdiff <= hysteresis_degc) {
          timer = now + SEC_TO_MS(period_seconds);
          break;
        }
        else if (PENDING(now, timer)) break;
        state = TRRunaway;

      } // fall through

      case TRRunaway:
        TERN_(HAS_DWIN_E3V2_BASIC, DWIN_Popup_Temperature(0));
        _temp_error(heater_id, FPSTR(str_t_thermal_runaway), GET_TEXT_F(MSG_THERMAL_RUNAWAY));

      #if ENABLED(THERMAL_PROTECTION_VARIANCE_MONITOR)
        case TRMalfunction:
          TERN_(HAS_DWIN_E3V2_BASIC, DWIN_Popup_Temperature(0));
          _temp_error(heater_id, FPSTR(str_t_temp_malfunction), GET_TEXT_F(MSG_TEMP_MALFUNCTION));
      #endif
    }
  }

  #pragma GCC diagnostic pop

#endif // HAS_THERMAL_PROTECTION
```

and:

```
++#pragma GCC diagnostic push
++#pragma GCC diagnostic ignored "-Wimplicit-fallthrough"
switch (adc_sensor_state) {

    --#pragma GCC diagnostic push
    --#if __has_cpp_attribute(fallthrough)
      --#pragma GCC diagnostic ignored "-Wimplicit-fallthrough"
    --#endif

    case SensorsReady: {
      // All sensors have been read. Stay in this state for a few
      // ISRs to save on calls to temp update/checking code below.
      constexpr int8_t extra_loops = MIN_ADC_ISR_LOOPS - (int8_t)SensorsReady;
      static uint8_t delay_count = 0;
      if (extra_loops > 0) {
        if (delay_count == 0) delay_count = extra_loops;  // Init this delay
        if (--delay_count)                                // While delaying...
          next_sensor_state = SensorsReady;               // retain this state (else, next state will be 0)
        break;
      }
      else {
        adc_sensor_state = StartSampling;                 // Fall-through to start sampling
        next_sensor_state = (ADCSensorState)(int(StartSampling) + 1);
      }
    }

    #pragma GCC diagnostic pop
```

### Version of Marlin Firmware

2.1.2.2

### Electronics

Arduino Due + Ramps-FD

### LCD/Controller

RepRapDiscount Smart Controller

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

mh-alinaghipour, 2024-04-04:
I just downloaded the latest [bugfix-2.1.x](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) version and tested defining additional axes as rotational. New compiling errors occur:
```
\Marlin-bugfix-2.1.x\Marlin\src\lcd\HD44780\marlinui_HD44780.cpp: In static member function 'static void MenuItem_static::draw(uint8_t, FSTR_P, uint8_t, const char*)':
\Marlin-bugfix-2.1.x\Marlin\src\lcd\HD44780\marlinui_HD44780.cpp:1210:43: error: variable-sized object 'estr' may not be initialized
     char estr[calculateWidth(ftpl) + 3] = "\0";
                                           ^
exit status 1
Compilation error: variable-sized object 'estr' may not be initialized
```
I was able to fix the above by implementing these changes to `"\Marlin-bugfix-2.1.x\Marlin\src\lcd\HD44780\marlinui_HD44780.cpp"`, (line 1210):

```
--char estr[calculateWidth(ftpl) + 3] = "\0";
++char estr[calculateWidth(ftpl) + 3];
++memset(estr, 0, (calculateWidth(ftpl) + 3)*(calculateWidth(ftpl) + 3)*sizeof(char));
```

But after that, the exact same pragma GCC diagnostics errors occur in `"\Marlin-bugfix-2.1.x\\Marlin\src\module\temperature.cpp"` (lines 3103 & 4117):

```
\Marlin-bugfix-2.1.x\Marlin\src\module\temperature.cpp:3103:26: error: missing binary operator before token "("
   #if __has_cpp_attribute(fallthrough)
                          ^
\Marlin-bugfix-2.1.x\Marlin\src\module\temperature.cpp:4117:28: error: missing binary operator before token "("
     #if __has_cpp_attribute(fallthrough)
                            ^

exit status 1

Compilation error: missing binary operator before token "("
```

which I was able to fix by doing the same changes to suppress pragma GCC diagnostics errors in the previous comment.

thisiskeithb, 2024-04-04:
Please attach your `bugfix-2.1.x` configs.

mh-alinaghipour, 2024-04-04:
> Please attach your `bugfix-2.1.x` configs.

bugfix-2.1.x Configs:
https://github.com/MarlinFirmware/Marlin/files/14865695/Marlin-bugfix-2.1.x-config.zip

thisiskeithb, 2024-04-04:
How are you compiling your build? Compiling works fine under macOS with VSCode + PlatformIO:

```prolog
Processing DUE (platform: atmelsam; board: due; framework: arduino)
-----------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/due.html
PLATFORM: Atmel SAM (8.2.0) > Arduino Due (Programming Port)
HARDWARE: AT91SAM3X8E 84MHz, 96KB RAM, 512KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduino-sam @ 1.6.12 
 - framework-cmsis @ 1.40500.0 (4.5.0) 
 - framework-cmsis-atmel @ 1.2.2 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 4 compatible libraries
Scanning dependencies...
Dependency Graph
|-- LiquidCrystal @ 1.5.1
|-- Wire @ 1.0
|-- SPI @ 1.0
Building in release mode
```
...
```prolog
Archiving .pio/build/DUE/libFrameworkArduino.a
Indexing .pio/build/DUE/libFrameworkArduino.a
Linking .pio/build/DUE/firmware.elf
Checking size .pio/build/DUE/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  11.5% (used 11280 bytes from 98304 bytes)
Flash: [====      ]  43.9% (used 230184 bytes from 524288 bytes)
Building .pio/build/DUE/firmware.bin
============================= [SUCCESS] Took 22.12 seconds =============================

Environment    Status    Duration
-------------  --------  ------------
DUE            SUCCESS   00:00:22.117
============================== 1 succeeded in 00:00:22.117 ==============================
 *  Terminal will be reused by tasks, press any key to close it. 
```

mh-alinaghipour, 2024-04-04:
I compile(verify) and upload Marlin onto my Arduino Due board with Arduino IDE 2.3.2 on Windows 11.

thisiskeithb, 2024-04-04:
> I compile(verify) and upload Marlin onto my Arduino Due board with Arduino IDE 2.3.2 on Windows 11.

Please use VSCode with PlatformIO instead: [Link](https://marlinfw.org/docs/basics/install_platformio_vscode.html).

Also, [from our README](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.1.x?tab=readme-ov-file#building-marlin-21):

> You can still build Marlin with Arduino IDE, and we hope to improve the Arduino build experience, but at this time PlatformIO is the better choice.

mh-alinaghipour, 2024-04-05:
> Please use VSCode with PlatformIO instead: [Link](https://marlinfw.org/docs/basics/install_platformio_vscode.html).
>
> Also, [from our [...]

Thank you. I installed PlatformIO on my VSCode and managed to compile without errors.

closed 2024-07-20 by github-actions[bot]