MarlinFirmware/Marlin #26938: [FR] Z Probe Wizard to include move 0.01 as a option

Russkil, 2024-04-05:
### Is your feature request related to a problem? Please describe.

not directly a problem just allowing you to be more accurate to the value allowed to be set in the preset z offset value

### Are you looking for hardware support?

no

### Describe the feature you want

On the Z Probe wizard either add the option Move 0.01mm or change Move 0.025mm to Move 0.01mm to allow the setting to be closer to the option you can save to the eeprom.

ellensp, 2024-04-05:
Presuming your talking about PROBE_OFFSET_WIZARD and a marlin controlled LCD (vs those with their own firmware)

This already exists

```cpp
  if ((FINE_MANUAL_MOVE) > 0.0f && (FINE_MANUAL_MOVE) < 0.1f)
    SUBMENU_f(F(STRINGIFY(FINE_MANUAL_MOVE)), MSG_MOVE_N_MM, []{ _goto_manual_move_z(float(FINE_MANUAL_MOVE)); });

```

so you just need to enable it 

In Configuration_advanced 

```
#if HAS_MANUAL_MOVE_MENU
  #define MANUAL_FEEDRATE { 50*60, 50*60, 4*60, 2*60 } // (mm/min) Feedrates for manual moves along X, Y, Z, E from panel
  #define FINE_MANUAL_MOVE 0.025    // (mm) Smallest manual move (< 0.1mm) applying to Z on most machines

```

HAS_MANUAL_MOVE_MENU is enabled on displays that marlin can directly control/

ellensp, 2024-04-05:
And the code has just been verified to work as intended.

closed 2024-04-05 by ellensp