MarlinFirmware/Marlin #26939: [BUG] **schema.py** does not export schema.xx configs

classicrocker883, 2024-04-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm trying to export my config using `#define CONFIG_EXPORT` in **Configuration_adv.h**, using parameters `[1:'JSON', 2:'config.ini', 3:'schema.json', 4:'schema.yml']`

`1` and `2` work fine. However, `schema.json` and `schema.yml` are MIA. (they _should_ be in **.pio/build/...**)

looking in **buildroot\share\PlatformIO\scripts\schema.py**, it says "This script can also be run standalone from within the Marlin repo to generate JSON and YAML schema files."

So in trying that... I get:
>Error: 'charmap' codec can't decode byte 0x90 in position 117: character maps to <undefined>

but then trying it in a different terminal bash I get the same message except it says ` in position 6641:`

not sure what is going on here, I've updated my python and pip modules.

### Expected behavior

I suppose there should be a file called `schema.json` or `schema.yml`

### Actual behavior

no file appears, and error given when trying standalone script.

### Steps to Reproduce

1. use `#define CONFIG_EXPORT` `3` or `4` in **Configuration_adv.h**
2. look inside the **.pio/build/[env]** dir

part B:
1. in terminal type 
```bash 
python3 ./buildroot/share/PlatformIO/scripts/schema.py
```

### Version of Marlin Firmware

bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-04-06:
No config's no service!

ellensp, 2024-04-06:
/

closed 2024-04-06 by ellensp

thisiskeithb, 2024-04-06:
Cannot reproduce either, so I’m going to assume it’s a malformed config.

No configs were provided, so we’ll never know 🤷‍♂️

@classicrocker883: We shouldn’t have to ask you for configs every time you open a bug report. Please correct this habit and follow the template by provide everything that is requested, **including your configuration files**.