MarlinFirmware/Marlin #26940: [BUG] MSG_ATTACH_SD_MEDIA not defined in 2.1.2.2

ellensp, 2024-04-06:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

  MSG_ATTACH_USB_MEDIA is used by 2.1.2.2 but not in the code 

Marlin\src\lcd\menu\menu_main.cpp:424:25: error: 'MSG_ATTACH_SD_MEDIA' is not a member of 'Language_en'; did you mean 'MSG_ATTACH_USB_MEDIA'?

Needs this line added to Marlin/src/lcd/language)
/language_en.h
  LSTR MSG_ATTACH_USB_MEDIA               = _UxGT("Attach USB Drive");

### Bug Timeline

Since 2.1.2.2 was released

### Expected behavior

It should build

### Actual behavior

Marlin\src\lcd\menu\menu_main.cpp:424:25: error: 'MSG_ATTACH_SD_MEDIA' is not a member of 'Language_en'; did you mean 'MSG_ATTACH_USB_MEDIA'?

### Version of Marlin Firmware

2.1.2.2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-04-06:
just added for the record

closed 2024-04-06 by ellensp

thisiskeithb, 2024-04-27:
Patch for `2.1.x` submitted:
- https://github.com/MarlinFirmware/Marlin/pull/27020