MarlinFirmware/Marlin #26941: [BUGS + Solution] SingleNozzle compilation failed (Since introduction of active extruders arrows)

studiodyne, 2024-04-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

- Impossible compilation when single nozzle cfg . Since introduction of ' active extruders arrows'
- Compiler stops about ' missing bitmap'
- If user choose ' STATUS_COMBINE_HEATERS ' to solve this , then no custom_status_screen , no animated hotend , no animated bed and more bugs

Solution ;
In **src/lcd/dogm/status_sceen_DOGM.cpp line 326**
`#if HAS_MULTI_EXTRUDER && DISABLED(SLIM_LCD_MENUS)
        if (active_extruder == heater_id)
          u8g.drawBitmapP(_MAX(0, STATUS_HOTEND_X(heater_id) - 6), STATUS_HEATERS_Y + 3, 1, 5, status_active_extruder_indicator_bmp);
      #endif`

- Preprocessor conditionals is wrong , because singlenozzle have multi extruders , but only one hotend and, impossible to draw the 2x arrows that indicates active extruders. I wanted to make a PR, but, I prefer that the author of this recent code, take a look and verify if no breaks elsewhere.
- Works fine if I comment this block, all is solved
- Don't forget the ' STATUS_COMBINE_HEATERS' that fails with singlenozzle

Thks

### Bug Timeline

New

### Version of Marlin Firmware

2.1 bugfix

### Printer model

fancy single nozzle

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2024-04-06:
Duplicate of #26833

thisiskeithb, 2024-04-06:
This will be fixed in https://github.com/MarlinFirmware/Marlin/pull/26834

closed 2024-04-06 by thisiskeithb

studiodyne, 2024-04-06:
Thks . I go to see the fix

studiodyne, 2024-04-06:
@thisiskeithb  
I noticed that STATUS_COMBINE_HEATERS, with singlenozzle makes unexpected behaviors; or big bugs , if someone have time to make a sanity line. Because most users define this , by thinking it's required in singlenozzle cfg

Thks