MarlinFirmware/Marlin #26950: compiling marlin for CR-10 MAX

Blisk, 2024-04-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am try to compile marlin with marlin bugfix 2.1.x and configuration file for CR 10 MAX.
I get this error
PS K:\Downloads\Marlin-bugfix-2.1.x(1)\Marlin-bugfix-2.1.x> platformio run --silent -e mega2560
In file included from Marlin\src\lcd\extui\dgus_reloaded\../../../inc/MarlinConfigPre.h:37:0,
                 from Marlin\src\lcd\extui\dgus_reloaded\DGUSRxHandler.cpp:23:
Marlin\src\lcd\extui\dgus_reloaded\../../../inc/../core/macros.h: In instantiation of 'constexpr void LIMIT(V&, N1, N2) [with V = unsigned int; N1 = int; N2 = int]':
Marlin\src\lcd\extui\dgus_reloaded\DGUSRxHandler.cpp:814:14:   required from here
Marlin\src\lcd\extui\dgus_reloaded\../../../inc/../core/macros.h:109:12: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]        
     if (n1 > v) v = n1;
         ~~~^~~
Marlin\src\lcd\extui\dgus_reloaded\../../../inc/../core/macros.h:110:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]        
     else if (n2 < v) v = n2;
              ~~~^~~
In file included from Marlin\src\lcd\fontutils.cpp:39:0:
Marlin\src\lcd\fontutils.h: In function 'const char* get_utf8_value_cb(const char*, read_byte_cb_t, lchar_t&)':
Marlin\src\lcd\fontutils.h:68:20: error: redefinition of 'const char* get_utf8_value_cb(const char*, read_byte_cb_t, lchar_t&)'
 inline const char* get_utf8_value_cb(const char *pstart, read_byte_cb_t cb_read_byte, lchar_t &pval) {
                    ^~~~~~~~~~~~~~~~~
In file included from Marlin\src\lcd\../inc/../core/mstring.h:45:0,
                 from Marlin\src\lcd\../inc/MarlinConfig.h:61,
                 from Marlin\src\lcd\fontutils.cpp:32:
Marlin\src\lcd\../inc/../core/../lcd/utf8.h:68:20: note: 'const char* get_utf8_value_cb(const char*, read_byte_cb_t, lchar_t&)' previously defined here
 inline const char* get_utf8_value_cb(const char *pstart, read_byte_cb_t cb_read_byte, lchar_t &pval) {
                    ^~~~~~~~~~~~~~~~~
In file included from Marlin\src\lcd\fontutils.cpp:39:0:
Marlin\src\lcd\fontutils.h: In function 'uint8_t utf8_strlen(FSTR_P)':
Marlin\src\lcd\fontutils.h:75:16: error: redefinition of 'uint8_t utf8_strlen(FSTR_P)'
 inline uint8_t utf8_strlen(FSTR_P fstart) { return utf8_strlen_P(FTOP(fstart)); }
                ^~~~~~~~~~~
In file included from Marlin\src\lcd\../inc/../core/mstring.h:45:0,
                 from Marlin\src\lcd\../inc/MarlinConfig.h:61,
Marlin\src\lcd\../inc/../core/../lcd/utf8.h:75:16: note: 'uint8_t utf8_strlen(FSTR_P)' previously defined here
 inline uint8_t utf8_strlen(FSTR_P fstart) { return utf8_strlen_P(FTOP(fstart)); }
                ^~~~~~~~~~~
*** [.pio\build\mega2560\src\src\lcd\fontutils.cpp.o] Error 1
================================================================================================================================ [FAILED] Took 20.31 seconds ================================================================================================================================

Environment    Status    Duration
-------------  --------  ------------
mega2560       FAILED    00:00:20.311
=========================================================================================================================== 1 failed, 0 succeeded in 00:00:20.311 ===========================================================================================================================

### Version of Marlin Firmware

marlin bugfix 2.1.X

### Printer model

Creality CR-10 MAX

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/14903111/configuration.zip)

ellensp, 2024-04-08:
Cannot reproduce this error with official Marlin bugfix 2.1.x and these example config files

ellensp, 2024-04-08:
the file Marlin\src\lcd\fontutils.h does not exist in bugfix.2.1.x

You have probably extracted bugfix over and old version of marlin, so you have left over old files.
Do not do this.
Always extract into an empty directory

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2024-04-08 by ellensp

Blisk, 2024-04-08:
thank you I stared all over and unzip to new folder and add configuration files.
Now compiling works but I get firmware.hex file instead of firmware.bin
It is Atmega 2560 board.

ellensp, 2024-04-08:
Atmega 2560 expect .hex files not .bin files

Blisk, 2024-04-08:
thank you I just read about that I will manage to update.