MarlinFirmware/Marlin #26953: Marlin G34 improvement

MICHASCHWARZ, 2024-04-09:
### Is your feature request related to a problem? Please describe.

The automatic Z alignment G34 is not working well, so I would have an idea on how to improve this feature.

### Are you looking for hardware support?

I have an Octopus MAX EZ V1.0 board with Marlin 2.1.2 and 4 Z-axes.

### Describe the feature you want

First of all, it is necessary to measure 4 points on the bed to calculate the positions of each Z-axis. This is not easy unless you assume some boundary conditions. Instead of a sloped surface rotated by an x and y interpolation, it would be much easier to define that the measurement points should be on the transverse axes of +X+Y and -X-Y. The point in the middle of this with the z-height of the average of the 4 z-values of the measuring point defines the actual height. It is now very easy to calculate the z-positions and the exact correction value of each z-axis using the ray theorem. The distance from the center point to the measuring point and the difference of the z-values of these two points is the smaller triangle and the ratio between the distance of the z-axis to the center point divided by the distance of the measuring point to the center point is multiplied by the difference of the z-values of the center point and the measuring point, which gives the actual z-position. The inverse of this value is the exact correction value for this axis. When this is calculated for each axis, the bed is at all points of the height of the center point and the alignment is complete. This should be verified by an additional measurement to check the accuracy.

Now I need someone who can program this function/routine for Marlin. Maybe as G34 A?

sjasonsmith, 2024-04-09:
When 4-axis was added, I believe that it basically performs the same iterative correction approach that is used for 2-axis.

Years ago I added the option for 3-axis to specify known stepper-motor coordinates so that you could probe once and calculate the adjustments needed to rotate the plane to the correct position. This was able to take the time to level the bed on a particularly poorly-designed printer I used from 20+ iterations to 2, with the second iteration barely adjusting anything.

The configs suggest this feature also works with 4-axis, but I don't believe that to be true. It works by rotating the bed, and assumes that it is a flat surface. With 4 Z motors it would just rotate the warped surface in space. You would converge quickly, but without actually making the surface as flat as possible.

I believe the method you described matches what I have been thinking _might_ work for 4 axis. I am no mechanical engineer so didn't know whether the materials would actually behave predictably enough to allow this to work. Even if it did not work perfectly, I'm sure it would converge much faster than the current mechanism.

MICHASCHWARZ, 2024-04-09:
My bed is an 8mm ESG glass  with a flatness less than 13 µm. This glass is not fixed to the z axis so that an deformation of the glass is excluded. The glass bed lay on 4 big rubber grommets which are fixed on the z axis. These provide the dimensional compensation and, due to the static friction caused by the large mass, ensure that the glass remains in the XY position.
Yes, it is possible that only 3 of 4 z axis has contact to the glass. 
So it is very improtant to ensure that all 4 z axis have contact in front of the alignment.

MICHASCHWARZ, 2024-04-09:
[image]

InsanityAutomation, 2024-04-09:
When I originally wrote quad z support, a company had offered a machine for testing. I did what exists as a draft with no hardware on hand... It apparently was good enough for them as they vanished and never provided hardware to test with... So with no way to test, I had no method to debug any improvements or drive remaining to do so.

Can you write out 2 or 3 example measurements and z motor movements to use as a sanity check?

MICHASCHWARZ, 2024-04-09:
At moment my USB socket is broken and i can't export the log file as .txt. But i can take some pictures. (new board already ordered) I've recognized that first the approximation works but at some point it becomes worse again. 
To calculate the corretion values is much faster than an approximation and works for all quantities between 2 an n Z axis. An uneveness is not under this configuration because this will be handled with auto bed leveling by measuring the mesh. So under consideration that the bed is a perfect surface and the measuring points are on the cross diagonals of the z axis in the same distance to the Center the calculation is much easier, faster and more exact.

MICHASCHWARZ, 2024-04-10:
Here the requested data from my machine 
z_stepper_orientation 0
nozzle_probe_offset {28, 0, -19}
z_stepper_align_xy {{62,46},{328,46},{328,344},{62,344}}
G34_max_grade 8
Inerations 10
align_acc 0.01

1. Interation: 
Z2-Z1=1.22 Z3-Z2=0.95 Z3-Z1=2.17 Z4-Z3=1.69 Z4-Z2=0.74 Z4-Z1=0.49
action: 1:2=1.225 3-2=0.949 3-1=2.174 4-3=1.686 4-2=0.737 4-1=0.284
2. Interation: 
Z2-Z1=0.40 Z3-Z2=0.58 Z3-Z1=0.98 Z4-Z3=0.72 Z4-Z2=0.14 Z4-Z1=0.26
action: 1:2=0,398 3-2=0.582 3-1=0.980 4-3=0.719 4-2=0.137 4-1=0.261
3. Interation:
Z2-Z1=0.27 Z3-Z2=0.01 Z3-Z1=0.28 Z4-Z3=0.03 Z4-Z2=0.02 Z4-Z1=0.25
action: 1:2=0.266 3-2=0.013 3-1=0.279 4-3=0.030 4-2=0.017 4-1=0.249
4. Interation:
Z2-Z1=0.09 Z3-Z2=0.10 Z3-Z1=0.00 Z4-Z3=0.04 Z4-Z2=0.06 Z4-Z1=0.03
action: 1:2=0.094 3-2=0.099 3-1=0.005 4-3=0.040 4-2=0.059 4-1=0.035
Decreasing accuracy detected!
5. Interation:
Z2-Z1=0.23 Z3-Z2=0.09 Z3-Z1=0.14 Z4-Z3=0.01 Z4-Z2=0.08 Z4-Z1=0.15
action: 1:2=0.234 3-2=0.095 3-1=0.140 4-3=0.010 4-2=0.085 4-1=0.150
Decreasing accuracy detected!
6. Interation:
Z2-Z1=0.10 Z3-Z2=0.18 Z3-Z1=0.08 Z4-Z3=0.04 Z4-Z2=0.22 Z4-Z1=0.12
action: 1:2=0.099 3-2=0.179 3-1=0.081 4-3=0.041 4-2=0.220 4-1=0.121
Decreasing accuracy detected!
7. Interation:
Z2-Z1=0.29 Z3-Z2=0.09 Z3-Z1=0.19 Z4-Z3=0.17 Z4-Z2=0.27 Z4-Z1=0.02
action: 1:2=0.286 3-2=0.094 3-1=0.192 4-3=0.174 4-2=0.268 4-1=0.017
8. Interation:
Z2-Z1=0.03 Z3-Z2=0.06 Z3-Z1=0.04 Z4-Z3=0.09 Z4-Z2=0.16 Z4-Z1=0.13
action: 1:2=0.027 3-2=0.062 3-1=0.036 4-3=0.094 4-2=0.156 4-1=0.130
9. Interation:
Z2-Z1=0.11 Z3-Z2=0.07 Z3-Z1=0.04 Z4-Z3=0.02 Z4-Z2=0.04 Z4-Z1=0.07
action: 1:2=0.111 3-2=0.067 3-1=0.043 4-3=0.022 4-2=0.045 4-1=0.066
Decreasing accuracy detected!
10. Interation:
Z2-Z1=0.10 Z3-Z2=0.08 Z3-Z1=0.02 Z4-Z3=0.06 Z4-Z2=0.14 Z4-Z1=0.04
action: 1:2=0.096 3-2=0.079 3-1=0.017 4-3=0.060 4-2=0.139 4-1=0.043
Decreasing accuracy detected!

ACC: 0.14