MarlinFirmware/Marlin #26954: [BUG] Not possible to define J axis without I axis

hm2dev, 2024-04-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When defining a configuration of a printer/cnc with an x, y, z and j axis, but no i axis (e.g. because the rotary axis is parallel to y and not to x), compiling fails because 

a) a typo in  https://github.com/MarlinFirmware/Marlin/blob/311bfc99f3f47292f5d4d5e80f29be3de58071a7/Marlin/src/pins/pins_postprocess.h#L1250 
which I assume should be `#if J_E_INDEX >= MAX_E_STEPPERS || !PIN_EXISTS(J_STEP)` instead.
b)  the way the code in 
https://github.com/MarlinFirmware/Marlin/blob/311bfc99f3f47292f5d4d5e80f29be3de58071a7/Marlin/src/inc/Conditionals_LCD.h#L243 and following makes assumptions of the number of axes.
https://github.com/MarlinFirmware/Marlin/blob/311bfc99f3f47292f5d4d5e80f29be3de58071a7/Marlin/src/inc/Conditionals_LCD.h#L251

Compilation fails with 

    Marlin\src\pins\pins_postprocess.h:1166:8: error: #error "No E stepper plug left for I!"
        #error "No E stepper plug left for I!"

### Bug Timeline

unknown

### Expected behavior

Compilation should succeed with only J and no I axis.

### Actual behavior

Comilation fails, with 

    Marlin\src\pins\pins_postprocess.h:1166:8: error: #error "No E stepper plug left for I!"
       #error "No E stepper plug left for I!"

### Steps to Reproduce

1. Define a configuration with no i axis and with j axis.
2. Compile

### Version of Marlin Firmware

2.1.x

### Printer model

not relevant. Example Board: MINIRAMBO_CNC

### Electronics

MINIRAMBO_CNC

### LCD/Controller

ULTIPANEL

### Other add-ons

none

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/14923437/Configuration_adv.zip)

hm2dev, 2024-04-09:
Upon further investigating and thinking about the axes b) might be working as designed and can be "fixed" by renaming the I axis to B in configuration.h instead of defining an J axis without an I axis.

https://github.com/MarlinFirmware/Marlin/blob/311bfc99f3f47292f5d4d5e80f29be3de58071a7/Marlin/Configuration.h#L183

This leaves the typo from a) as bug.

If that is so, I can provide a pull request fixing the typo.

ellensp, 2024-04-09:
Yes in Marlin the internal axis names must be used sequentially. You then give it the external name you want.

Yes that is a typo you found.

ellensp, 2024-04-17:
A pr has been created to fix this typo.

closed 2024-04-17 by ellensp