MarlinFirmware/Marlin #26961: [BUG] Marlin 2.1.2.2 reboots on startup on the Sovol SV06

mlewis-everley, 2024-04-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am trying to run the latest Marlin (and also setup a version that will allow me easily merge in updates from upstream) for the Sovol SV06.

I have first tried rebasing [this](https://github.com/blastrock/Marlin-sv06) version against the latest `2.1.x` branch. Once I flashed the firmware to the latest Marlin build, the:

1. Printer boots
2. Displays the "Marlin 2.1.2.2" boot screen
3. Reboots (GoTo 2.)

In case this was an issue with the rebase, I tried manually copying the configuration.h and configuration_adv.h from the blastrock version into a vanilla Marlin install, but also get the same results.

I have also tried flashing the blastrock binary to the printer and that installs fine and the printer boots.

I noticed while performing the rebase that the latest version of marlin uses the `GD32` processor for the Sovol board, where as the "Official" or blastrock version use an `STM32` processor. Could it be something related to that?

### Bug Timeline

New?

### Expected behavior

After flashing the firmware, the Sovol SV06 will boot into Marlin 2.1.2.2

### Actual behavior

Printer is stuck in a boot/reboot loop

### Version of Marlin Firmware

2.1.2.2 (2.1.x branch)

### Printer model

Sovol SV06

### Electronics

Stock

### LCD/Controller

CR10/12864 display

### Other add-ons

None

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/14952444/configuration.zip)

thisiskeithb, 2024-04-12:
Duplicate of #26811

thisiskeithb, 2024-04-12:
This issue doesn't exist in [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip), so you'll need to disable `INPUT_SHAPING_X`/`INPUT_SHAPING_Y` or move to the latest `bugfix-2.1.x`.

closed 2024-04-12 by thisiskeithb

mlewis-everley, 2024-04-12:
> This issue doesn't exist in [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip), so you'll need to disable `INPUT_SHAPING_X`/`INPUT_SHAPING_Y` or move to the latest `bugfix-2.1.x`.

@thisiskeithb  This is not what is happening to me. If I disable input shaping I get an "Err: EEPROM Version" message, but if I select either "Reset" or "Ignore", the printer reboots and throws the same error.

If I switch to the latest bugfix-2.1.x with input shaping enabled, I also get this error and boot loop.

mlewis-everley, 2024-04-14:
@thisiskeithb I have tried this again today, I cannot get the latest Marlin `bugfix-2.1.x` release to work on the Sovol SV06.

I have tried the bugfix version with input shaping enabled and disabled. I have also tried changing other settings in config that I have changed (such as stepper disabling and stepper current) and this has also had no effect.

Finally, I have line for line transcribed the blastrock version (linked in original report) configuration files onto a new install of Marlin `bugfix-2.1.x` and still get stuck in a reboot loop.

To clarify, this issue does not appear to be related to input shaping. The only thing I can see that has changed from the official/blastrock forks of Marlin is that the latest version has migrated from using `STM32F103RET6_sovol_maple` to `GD32F103RET6_sovol_maple`

I have even attempted to change back to using an STM based configuration, but unfortunately Marlin will no longer allow this for `BOARD_SOVOL_V131` and I an not familiar enough with Marlin to change this deeper in the codebase.

mlewis-everley, 2024-04-18:
Sooooo, apparently no one wants to re-open this issue, but, I have done some more work on this, looks like the pin mapping for `BEEPER_PIN` in `pins_CREALITY_V4.h` is not compatible with the SV06.

Exact line here: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/pins/stm32f1/pins_CREALITY_V4.h#L241

This appears to be an issue in Marlin 2.1.x globally? If you use the stock firmware (based on Marlin 2.0.x) the beeper works fine (also mapped to `PC6`).

If I set `BEEPER_PIN -1` in the config file, then I can flash the latest Marlin version from `bugfix-2.1.x` and the printer runs fine.

As no one is willing to re-open this issue, I guess I will have to create a new one (if one related to this exact issue does not exist).