MarlinFirmware/Marlin #26966: [FR] Debugging for other boards/environments

classicrocker883, 2024-04-13:
### Are you looking for hardware support?

it appears that Debugging is only released to one thing, simulator.

It would be cool having support for everything else. 

### Describe the feature you want

introduced support for Debugging more than one thing. 

### Additional context

Q: is there a way in VS Code to already do Debugging that I'm not aware of?

sjasonsmith, 2024-04-13:
It is definitely possible to debug on real hardware, but it typically requires a JTAG interface. The actual requirements vary by board and MCU. It is sometimes very easy, and sometimes impossible.

Some environments have debug information configured in them (or defaulted in PlatformIO), but they are going to reflect whatever a developer had used on them previously, and won't necessarily match what you have on hand.

sjasonsmith, 2024-04-13:
There probably aren't really Marlin-specific instructions for this, but you can find information about how to configure debugging with PlatformIO to get started.

thisiskeithb, 2024-04-13:
Here's an STM32F1 example:

# Debug Marlin 2.x with VSCode & PlatformIO using an ST-Link V2 (STM32F1)

## Table of Contents

- [Connect ST-Link V2 to SWD Pins](#connect-st-link-v2-to-swd-pins)
- [Configure Marlin](#configure-marlin)
- [Configure & Run GDB](#configure--run-gdb)
  * [macOS](#macos)
  * [Windows](#windows)
- [Basic GDB Debugging Commands](#basic-gdb-debugging-commands)

## Connect ST-Link V2 to SWD Pins

This will vary depending on the board.

## Configure Marlin

Using your desired config settings, you will also need to:

1. Disable `USE_WATCHDOG`

2. Enable `MARLIN_DEV_MODE`

3. Possibly disable `DISABLE_DEBUG` and/or `DISABLE_JTAG` depending on your board (check the board's pins file).

4. Build Marlin

## Configure & Run GDB

### macOS

1. In VSCode, open Terminal (<kbd>CTRL</kbd> + <kbd>`</kbd>) and run the following to configure OpenOCD (STM32F1):

   ```
   ~/.platformio/packages/tool-openocd/bin/openocd -d2 -s ~/.platformio/packages/tool-openocd/scripts -f interface/stlink.cfg  -f target/stm32f1x.cfg
   ```
    💡 See `~/.platformio/packages/tool-openocd/scripts/target` for a full list of `*.cfg` files 💡

2. Then open a new Terminal (<kbd>CTRL</kbd> + <kbd>SHIFT</kbd> + <kbd>`</kbd>) and run the following to launch GDB for debugging:

   ```
   ~/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-gdb /path/to/marlin/.pio/build/YOUR_PIO_ENV/firmware.elf
   ```

    💡 Be sure to change the path to `firmware.elf` in the command above to match your build environment. 💡

3. In the same Terminal window as Step 2, run the following command to connect GDB to the ST-Link:

   ```
   tar ext:3333
   ```
### Windows

1. In VSCode, open Terminal (<kbd>CTRL</kbd> + <kbd>`</kbd>) and run the following to configure OpenOCD (STM32F1):

   ```
   %USERPROFILE%\.platformio\packages\tool-openocd\bin\openocd -d2 -s %USERPROFILE%\.platformio\packages\tool-openocd\scripts -f interface\stlink.cfg  -f target\stm32f1x.cfg
   ```
    💡 See `%USERPROFILE%\.platformio\packages\tool-openocd\scripts\target` for a full list of `*.cfg` files 💡

2. Then open a new Terminal (<kbd>CTRL</kbd> + <kbd>SHIFT</kbd> + <kbd>`</kbd>) and run the following to launch GDB for debugging:

   ```
   %USERPROFILE%\.platformio\packages\toolchain-gccarmnoneeabi\bin\arm-none-eabi-gdb C:\path\to\marlin\.pio\build\YOUR_PIO_ENV\firmware.elf
   ```

    💡 Be sure to change the path to `firmware.elf` in the command above to match your build environment. 💡

3. In the same Terminal window as Step 2, run the following command to connect GDB to the ST-Link:

   ```
   tar ext:3333
   ```

## Basic GDB Debugging Commands

| Command   | Description                                                                              |
|:---------:|------------------------------------------------------------------------------------------|
| `bt`      | Backtrace. Print the current function stack to show where you are in the current program |
| `bt full` | Backtrace, including local variables                                                     |
| `b`       | Break on the first line in Marlin firmware                                               |
| `n`       | Step over                                                                                |
| `s`       | Step into                                                                                |
| `c`       | Continue                                                                                 |
| `q`       | Quit                                                                                     |
| `r`       | Restart                                                                                  |

closed 2024-04-13 by thisiskeithb