MarlinFirmware/Marlin #26970: [BUG] sd read error when starting print from lcd with color_ui

piegolf, 2024-04-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

when starting a print using the lcd, the lcd freezes and wont respond, connecting to the printer using pronterface shows "SD read error" when the print starts, issue does not show up using classic_ui instead of color_ui and/or when starting a print from sd card using pronterface

### Bug Timeline

i found the issue yesterday when installing marling using color ui

### Expected behavior

print starts

### Actual behavior

lcd freezes and wont respond
[image]

### Steps to Reproduce

1. compile marlin for mks_robin_nano, sdsupport, MKS_ROBIN_TFT_24, TFT_COLOR_UI, and TOUCH_SCREEN. 2. flash the board with the firmware. 3. attempt to start print from sd card using the lcd.

### Version of Marlin Firmware

2.1.2.2

### Printer model

Kingroon KP3S

### Electronics

stock, MKS_ROBIN_NANO and MKS_ROBIN_TFT_24

### LCD/Controller

MKS_ROBIN_TFT_24

### Bed Leveling

None

### Your Slicer

Other (explain below)

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/14972089/configs.zip)

thisiskeithb, 2024-04-14:
`2.1.2.2` has been a very buggy release. Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

piegolf, 2024-04-14:
the behavior is the same in the bugfix release
[image]

piegolf, 2024-07-20:
issue still exists with latest bugfix 2.1.x. latest bugfix 2.0.x didn't have any issues with a similar config and the same printer so its a 2.1.x specific issue

piegolf, 2024-09-01:
#27385 fixes issue on my printer

[image]

thinkyhead, 2024-09-03:
Good news that we're hitting on a solution. I was personally stumped. I'd like to try a few things to better understand how peripherals can conflict with each other. Please join us over at #27385 if you'd like to help us test some changes designed to make SD card I/O more resilient in a shared bus situation.

closed 2024-09-05 by thinkyhead