MarlinFirmware/Marlin #26971: [BUG] (MKS_GEN_L extruder heater not working)

Darkic5, 2024-04-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The extruder fan would not turn on when reaching or exceeding 50 degrees C. Tried old stable firmware I had (1.1.9), and it worked. After trying multiple configurations for 2.1.x and making sure all settings are correct, I compared the pins file for MKS_GEN_L and found that the wrong MOSFET was assigned to pin 7. Currently it sais #define MOSFET_B_PIN, but this should be #define MOSFET_D_PIN.
Changing to #define MOSFET_D_PIN and building the firmware again has fixed the issue for me! 
I suppose everyone using this board will face the same issues, so could you update this please?

### Bug Timeline

Around a year ago (same issue with old bugfix 2.1.x firmware I downloaded around a year ago)

### Expected behavior

Extruder fan to start spinning above 50 degrees C

### Actual behavior

Extruder fan never started spinning

### Steps to Reproduce

- Build firmware for MKS_GEN_L board.
- Use Artillery Genius BL touch configuration file
- Flash firmware
- Set extruder temperature to 60 degrees C 

### Version of Marlin Firmware

bugfix 2.1.x 13-4-2024

### Printer model

Artillery Genius

### Electronics

Stock + BL touch

### LCD/Controller

MKS TFT28 V1.0

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration bugfix 2.1.x 14-4-2024.zip](https://github.com/MarlinFirmware/Marlin/files/14972277/Configuration.bugfix.2.1.x.14-4-2024.zip)

thisiskeithb, 2024-04-16:
This was "fixed" in https://github.com/MarlinFirmware/Marlin/pull/25717 since @Nomis-97 was [following the schematic](https://github.com/makerbase-mks/MKS-GEN_L/blob/master/hardware/MKS%20Gen_L%20V1.0_008/MKS%20Gen_L%20V1.0_008%20SCH.pdf), but that doesn't match [the pins diagram](https://github.com/makerbase-mks/MKS-GEN_L/blob/master/hardware/MKS%20Gen_L%20V1.0_008/MKS%20Gen_L%20V1.0_008%20PIN.pdf) in pin numbers or pin counts/connector names and are likely typos in the schematic.

Pin 7 has been mapped to HE1/J10 (as marked in the pins diagram) for as long as I can remember using the Gen-L, so that PR should likely be reverted.

thisiskeithb, 2024-04-16:
I just traced the FAN (`FAN0_PIN`), HE1 (`HEATER_1_PIN`), HE0 (`HEATER_0_PIN`), and H-BED (`HEATER_BED_PIN`) pins to the MCU on my Gen-L V1.0 and can confirm that MKS' Gen-L V1.0 pins diagram is correct, not the schematic.

thisiskeithb, 2024-04-16:
PR submitted that should hopefully resolve MKS Gen-L V1 pin issues once & for all: https://github.com/MarlinFirmware/Marlin/pull/26974

closed 2024-08-11 by thinkyhead