MarlinFirmware/Marlin #26973: [FR] Add driver type for MKS SERVO42C motors

PanosPetrou, 2024-04-15:
### Is your feature request related to a problem? Please describe.

I have two MKS SERVO42C motors on X & Y axes of my printer (my board is an MKS Tinybee, running bugfix 2.1x). These are closed loop motors by MKS and they are using an integrated custom driver. From day one I had layer shifts, moving more steps than they should and pretty much causing problems.

However I manage to make them play nice using the following settings:

 On Configuration.h I set the following:
    #define X_DRIVER_TYPE A4988
    #define Y_DRIVER_TYPE A4988
    Disable S_CURVE_ACCELERATION (just in case, perhaps this is not needed)

On Configuration_adv.h I set the following:
    #define MIN_STEPS_PER_SEGMENT 12
    #define MINIMUM_STEPPER_POST_DIR_DELAY 1000
    #define MINIMUM_STEPPER_PRE_DIR_DELAY 1000
    #define MINIMUM_STEPPER_PULSE 4
    #define MAXIMUM_STEPPER_RATE 20000
    Disable ADAPTIVE_STEP_SMOOTHING (just in case, perhaps this is not needed)

Now the motors work reliably, no layer shifts after days of printing continuously. I think the important part is that this motor controller can't handle rates above 20KHz well.

So I was wondering if a driver type can be added in Marlin for these motors using the above settings. It would simplify the setup of these motors, using something like:
    #define X_DRIVER_TYPE SERVO42C
    #define Y_DRIVER_TYPE SERVO42C

### Are you looking for hardware support?

Yes, better support of MKS SERVO42C.

### Describe the feature you want

I was wondering if a driver type can be added in Marlin for these motors using the above settings. It would simplify the setup of these motors, using something like:
    #define X_DRIVER_TYPE SERVO42C
    #define Y_DRIVER_TYPE SERVO42C

thisiskeithb, 2024-06-09:
This would be something that MKS should add to to their SERVO42C documentation and not for us to add a workaround / force a fake driver driver type to set these values.

closed 2024-06-09 by thisiskeithb

PanosPetrou, 2024-06-09:
I understand, thank you for your time.