MarlinFirmware/Marlin #26982: [BUG] Using `BEEPER_PIN` in Marlin 2.1.x on the Sovol SV06 causes the printer to crash / boot loop

mlewis-everley, 2024-04-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

As I reported in issue #26961, My SV06's crash when trying to boot using Any stock branch of Marlin 2.1.x (including `bugfix-2.1.x`).

After a lot of mucking around, I have found that the issue is the `BEEPER_PIN` definition in: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/pins/stm32f1/pins_CREALITY_V4.h#L241

Disabling the beeper (`BEEPER_PIN -1`) allows the printer to boot, and it functions normally, but the speaker is really useful for flagging things such as filament runout/change/etc.

I have also looked around some of the other issues here, I can see that in #25868 (and the warning during compilation) that Maple environments are depreciated, but there is no non Maple alternative for the SV06?

I also spotted issue #20646 which outlines the need to an additional board on the Ender 3 Pro (similar but also different issue).

Finally, I am aware of PR: #25864 , but this is for the SV06 plus touchscreen.

Are there any solutions/possible fixes for the stock SV06? Is it an issue with the screen (would it just be easier to but a different one)?

### Bug Timeline

Seems to be present in any version 2.1.x

### Expected behavior

The printer boots as normal with speaker enabled.

When the printer boots (or the dial on the LCD interacted with), I hear an audible tone from speaker.

### Actual behavior

Enabling the speaker via `beeper_pin` causes the printer to crash on boot and re-boot.

### Steps to Reproduce

1. Download marlin `bugfix-2.1.x`
2. Configure marlin for the Sovol SV06 (using `GD32F103RET6_sovol_maple`) using a beeper
3. Compile using PlatformIO
4. Flash for Sovol SV06
5. Watch as printer constantly re-boots

### Version of Marlin Firmware

2.1.x/bugfix

### Printer model

Sovol SV06

### Electronics

Stock printer

### LCD/Controller

12864 LCD

### Other add-ons

None

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

My own fork of Marlin (using `bugfix-2.1.x`): https://github.com/mlewis-everley/Marlin/tree/2.1.x-SV06

configuration.h: https://github.com/mlewis-everley/Marlin/blob/2.1.x-SV06/Marlin/Configuration.h
configuration_adv.h: https://github.com/mlewis-everley/Marlin/blob/2.1.x-SV06/Marlin/Configuration_adv.h

thisiskeithb, 2024-04-28:
This appears to be a known issue with the SV06/GD32 motherboard: 

> The speaker is disabled in the configuration because of crash issues it causes with the clone processor used in the Sovol.  
> 
> But the speaker still works fine and produces the beeps and sounds you would expect. For those who want it disabled you can do so in the menu and save settings to not hear anything.

_Originally posted by @hillsoftware in https://github.com/hillsoftware/sv06/issues/13#issuecomment-1485548537_

mlewis-everley, 2024-04-28:
Hi @thisiskeithb, thank you for looking into this. I am not sure if it helps, but the stock fork of Marlin that Sovol created doesn't seem to have this issue, even though the pin mapping appears identical.

Is there anything about the Sovol fork that may be worth checking?

thisiskeithb, 2024-04-28:
> Is there anything about the Sovol fork that may be worth checking?

The [SV06 source code](https://github.com/Sovol3d/Sv06-Source-Code) appears to be based off Marlin `2.0.9.2`, so I diffed them and and didn't see anything related to pins/speaker/timers that may be the culprit.

Sovol disabled `USE_WATCHDOG` in `Configuration_adv.h`, so as a **_temporary_** test, does disabling `USE_WATCHDOG` with `SPEAKER` enabled work?

mlewis-everley, 2024-04-29:
Thanks for looking at this @thisiskeithb . I am pretty stacked this week, but I will see if I can find some time try this out!

mlewis-everley, 2024-05-22:
OK, only just gotten around to looking at this! I attempted to turn off watchdog and restore the beeper pin and the printer wouldn't even boot when trying to flash the firmware (just got an infinite blank screen).

Experimentally, I re-enabled watchdog then left the beeper pin as `PC6` but then disabled `SPEAKER` in configuration.h. Flashing this now results in the speaker working!

I have no idea what is going on, it seems the issue is related to the `SPEAKER` config instead?