MarlinFirmware/Marlin #26986: Dual Y stepper not working

Carlsans, 2024-04-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

HAS_DUAL_Y_STEPPERS is not set on anywhere. Had to enable it in Conditionals_LCD that way : 
#ifdef Y2_DRIVER_TYPE
  #define HAS_DUAL_Y_STEPPERS 1
#endif
When enabled that way, the y axis work properly except when homing. I use : 
#define HOME_Y_BEFORE_X                    
#define CODEPENDENT_XY_HOMING 
the z axis move but then nothing on the y axis : 
Recv: Machine Type: Cartesian
Recv: Probe: FIX_MOUNTED_PROBE
Recv: Probe Offset X0.00 Y0.00 Z0.00 (Aligned With & Same Z as Nozzle)
Recv: Auto Bed Leveling: BILINEAR (disabled)
Recv: 
Recv: >>> set_bed_leveling_enabled  X-5.00 Y-20.00 Z0.00
Recv: <<< set_bed_leveling_enabled  X-5.00 Y-20.00 Z0.00
Recv: Raise Z before homing:
Recv: do_blocking_move_to_z(20.00, 4.00)
Recv: >>> do_blocking_move_to  X-5.00 Y-20.00 Z0.00
Recv: >  X-5.00 Y-20.00 Z20.00
[...]
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
[...]
Recv: echo:busy: processing
[...]
Recv: <<< do_blocking_move_to  X-5.00 Y-20.00 Z20.00
Recv: >>> homeaxis(Y)
Recv: Home Fast: -1357.50mm
Recv: >>> do_homing_move  X-5.00 Y-20.00 Z20.00
Recv: ...(Y, -1357.50, [50.00])

My machine is very unconventional and have a large bed : 

#define X_BED_SIZE 921
#define Y_BED_SIZE 885

### Version of Marlin Firmware

02010202

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[gitissue.zip](https://github.com/MarlinFirmware/Marlin/files/15045149/gitissue.zip)

thisiskeithb, 2024-04-19:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/26801

closed 2024-04-19 by thisiskeithb

Carlsans, 2024-04-19:
I've done exactly what was mentionned in https://github.com/MarlinFirmware/Marlin/issues/26801 but the homing issue is still present.

thisiskeithb, 2024-04-19:
> I've done exactly what was mentionned in #26801 but the homing issue is still present.

You don't need to change anything. `HAS_DUAL_Y_STEPPERS` no longer exists in current code, so download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) and port your config over.

Carlsans, 2024-04-19:
Thanks, I will do that returning from weekend. Really thank you for your time.
Carl