MarlinFirmware/Marlin #26989: [BUG] tmc2209 motor driver uart not working with skr pro v1.2

fyavuz96, 2024-04-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

With skr pro v1.2, tmc2209 motor driver uart does not work, but when I switch to the tmc2209 standalone version, it starts working. It was working normally in the old version marlin. When I upgrade to the latest version, none of the engines work.

### Expected behavior

I was expecting the motors to work with tmc2209 watch uart.

### Actual behavior

None of the motor drivers worked.

### Version of Marlin Firmware

Marlin-2.1.2.2

### Printer model

BTT skr pro V1.2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

fyavuz96, 2024-04-20:
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/15049195/Configuration_adv.zip)

ellensp, 2024-04-20:
You have enabled   #define SOFTWARE_DRIVER_ENABLE
But tmc2209 use a enable pin.

This stops all stepper drivers moving.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2024-04-20 by ellensp