MarlinFirmware/Marlin #26997: [BUG] Compilation error: MPCTEMP_START defined in dwin.h, but used in temperature.cpp when EXTENSIBLE_UI is enabled

rondlh, 2024-04-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

MPCTEMP_START is defined in dwin.h, but used in temperature.cpp when EXTENSIBLE_UI and MPC_AUTOTUNE are enabled, this causes a compilation error when you don't use DWIN_CREALITY_LCD.

Marlin\src\module\temperature.cpp: error: 'MPCTEMP_START' is not a member of 'ExtUI::mpcresult_t'
 1230 |     TERN_(EXTENSIBLE_UI, ExtUI::onMPCTuning(ExtUI::mpcresult_t::MPCTEMP_START));

### Bug Timeline

Recent update (last 2 weeks I guess)

### Expected behavior

No compilation error

### Actual behavior

Compilation error:
Marlin\src\module\temperature.cpp: error: 'MPCTEMP_START' is not a member of 'ExtUI::mpcresult_t'
 1230 |     TERN_(EXTENSIBLE_UI, ExtUI::onMPCTuning(ExtUI::mpcresult_t::MPCTEMP_START));

### Steps to Reproduce

1. Enable EXTENSIBLE_UI and MPC_AUTOTUNE
2. Compile the source

### Version of Marlin Firmware

Bugfix-2.1.x April 21 2024

### Printer model

Custom

### Electronics

MKS Monster8 V1.0

### LCD/Controller

BTT TFT35 V3.0

### Other add-ons

MKS EBB42 CAN Toolhead, BLTouch 3.1

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/15063523/Configuration_adv.zip)

ellensp, 2024-04-23:
The point of asking for config files is so we have something we can drop in and run to reproduce the issue.

Your configs do not do this, they are not usable, infact they reference code that is not even in stock marlin.

ellensp, 2024-04-23:
A simple config that will trigger the error and builds
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/15076106/Configuration.zip)

ellensp, 2024-04-23:
This PR fixes it

closed 2024-04-23 by ellensp

rondlh, 2024-04-23:
> The point of asking for config files is so we have something we can drop in and run to reproduce the issue.
> 
> Your configs do not do this, they are not usable, infact they reference UI code that is not even in stock marlin.

Sorry, I understand your point, will do better next time.