MarlinFirmware/Marlin #26999: Not sure if code is working with RAMPS 1.4 Shield + LCD

YoNoTacos4U, 2024-04-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I've just recently got a replica arduino mega, with a ramps 1.4 shield, a power supply and a LCD (RepRap 12864). I plugged everything in, and the LCD is just blue. I've already tried uploading the code multiple ways, same thing. I've also seen it is common for just switching the wires on the LCD, and I've tried that and still nothing. Not sure if anyone could help?

### Bug Timeline

Always like this since first time turning on

### Expected behavior

Marlin boot screen to show on LCD

### Actual behavior

Blue blank screen

### Steps to Reproduce

Plug everything in and turn it on

### Version of Marlin Firmware

2.1.2.2

### Printer model

N/A

### Electronics

 Arduino+RAMPs: RAMPS 1.4, 3D Printer RAMPS 1.4 Controller + R3 + A4988 with Heat Sink USB Calbe Jumper Kit

### LCD/Controller

LCD:HiLetgo 3D Printer Reprap Smart Controller 12864 LCD Display with Smart Controller Board for 3D Printer RAMPS 1.4 Reprap Mendel Prusa for Arduino 128x64 LCD Blue Color

### Other add-ons

N/A

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

PanosPetrou, 2024-04-22:
Keep in mind some people report quality issues with the particular display. Did you attach a zip file containing your Configuration.h & Configuration_adv.h files? I don't see it.

ellensp, 2024-04-23:
2.1.2.2 is buggy and should be avoided. will reboot constantly with input shaping enabled

ellensp, 2024-04-23:
no configs. no way to check anything or help you. closing.

closed 2024-04-23 by ellensp