MarlinFirmware/Marlin #27001: [BUG] ESP3D_WIFISUPPORT Not working with MKS TinyBee Motherboard

danvie4, 2024-04-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin fails to compile with multiple errors, include this one.
"Marlin/src/HAL/ESP32/../../inc/SanityCheck.h:3683:4: error: #error "WIFI_CUSTOM_COMMAND requires an ESP32 motherboard and WIFIS"

I'm using VSCode and Auto Build Marlin to compile the firmware. The issue exists on both this version of marlin and the bug fix branch as best as I can tell. 
[image]

I believe it compiles fine in marlin 2.0.9.7, but does not compile in Marlin-2.1.x, Marlin 2.1.2.2, or Marlin-bugfix-2.1.x.

### Steps to Reproduce

Set "default_envs = mks_tinybee" in platformio.ini file.
In Configuration.h, "#define MOTHERBOARD BOARD_MKS_TINYBEE"
In Configuration_adv.h, "#define ESP3D_WIFISUPPORT "

### Version of Marlin Firmware

Marlin-2.1.2.2

### Printer model

Anet ET4+ 

### Electronics

MKS TinyBee Motherboard

### LCD/Controller

MKS12864 V3

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin-2.1.2.2_Config.zip](https://github.com/MarlinFirmware/Marlin/files/15070598/Marlin-2.1.2.2_Config.zip)
I have changed many of the settings to try to match my printer, but I believe the issues is reproducible with no changes other than stated in the steps to reproduce box.

Hopefully that's enough info. 
Any help is greatly appreciated, thanks!

thisiskeithb, 2024-04-23:
Your configs are for `2.1.2.2`, not the latest `bugfix-2.1.x`.

This was fixed in the latest `bugfix-2.1.x` a couple weeks ago in https://github.com/MarlinFirmware/Marlin/commit/e6755450df46870c961f8d675815da1415d6436f

closed 2024-04-23 by thisiskeithb

danvie4, 2024-04-23:
@thisiskeithb My apologies, I thought I had tested the bugfix branch the other night and it didn't work, but I tried it again tonight, and I was wrong, sorry to waste your time, thanks for the quick response!