MarlinFirmware/Marlin #27004: [BUG] Strange behavior of the PWM fans on bugfix-2.1.x MKS Monster8 v2

wzaburko, 2024-04-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm trying to set up a new MKS Monster 8 motherboard with MKS_MINI_12864_V3. I previously used FLY407ZB on firmware 2.1.2.1 and had no issues. However, when adjusting the configs for the new board and using #define MKS_MINI_12864_V3, the display doesn't work (the motherboard functions correctly). But when using #define FYSETC_MINI_12864_2_1, the display works but the CF memory card slots don't. Also, it frequently loses connection with OctoPrint, sometimes even disconnecting without printing. So, I tried 2.1.2.2, and after successful installation, the board doesn't work - only the Marlin logo appears on the display, and there's no connection to OctoPrint, neither with #define MKS_MINI_12864_V3 nor with FYSETC_MINI_12864_2_1. I tried bugfix - the board, display, and CF cards work fine, but there's something strange with the PWM fans. They behave as if something is trying to turn them off every second. This happens on all three PWM Fans. I tried assigning different functions to these outputs - Hotend fan, blower, cooling, FAST_PWM_FAN_FREQUENCY off/on. It's the same in all configurations. Later, I noticed a slightly different behavior. You can see it in the attached videos.

https://drive.google.com/file/d/1pd7qVFSDFjl94BnoA-pWHWapWWTcHoVc/view?usp=drive_link
https://drive.google.com/file/d/1e9ZJbsQ6c6__DYHxaZNaTAxtkHpqz9HL/view?usp=drive_link

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

personal project

### Electronics

MKS Monster 8 v2

### LCD/Controller

MKS_MINI_12864_V3

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Conf.zip](https://github.com/MarlinFirmware/Marlin/files/15080242/Conf.zip)

rondlh, 2024-04-23:
I have the same problem.

In buildroot\share\PlatformIO\variants\Marlin_F4x7Vx\variant.h

#define TIMER_SERIAL          TIM5  // TIMER_SERIAL must be defined in this file

**Change to:**

#define TIMER_SERIAL          TIM4  // TIMER_SERIAL must be defined in this file

This resolves the issue for me.

thisiskeithb, 2024-04-23:
Most of the display issues should be resolved in https://github.com/MarlinFirmware/Marlin/pull/26988

wzaburko, 2024-04-23:
I recompiled the bugfix - I changed TIM5 to TIM4 and I see that the fans are working correctly, the display and memory card also work, although when inserting and removing the card, the display dims for 2-3 seconds - I haven't investigated the cause yet - but works. But other than that (after a short test), it seems that everything is okay. I haven't tested 2.1.2.2. Thank you very much for the quick help.  :)

closed 2024-08-06 by github-actions[bot]