MarlinFirmware/Marlin #27010: [BUG]Freeze_feature does not stop all motors

karabas2011, 2024-04-24:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Freeze feature stops print head motion but not extruder motor

### Bug Timeline

I never use this feature before

### Expected behavior

I expect stop all motors.

### Actual behavior

When printing from SDCARD  only X/Y motors stopped
When idle, If  I start extruding via menu  - it stopped!
So there is limit of how many motors can be stopped.

### Steps to Reproduce

Start prining from SDCARD
Press freeze button

### Version of Marlin Firmware

marlin 2.1.2.1

### Printer model

sparky

### Electronics

 SKR1.3

### LCD/Controller

MKS12864

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-04-24:
Please test the [`bugfix-2.1.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate 
further.

Please attach your configuration files.

karabas2011, 2024-04-26:
https://github.com/MarlinFirmware/Marlin/assets/2634978/ef5d3168-07b4-4d63-8534-10c689c21fcd

ellensp, 2024-04-26:
Please attach your configuration files.

Please test the [`bugfix-2.1.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

closed 2024-08-12 by thinkyhead

karabas2011, 2024-08-12:
Sorry for late. last release 2.1.2.4 seems to work correctly.