MarlinFirmware/Marlin #27011: [FR] Home Offset Here for a specific Axis

Elias-Ta, 2024-04-24:
### Describe the feature you want

I need to set Homing offset for a single axis in the current position, this can be usefull in cnc mode if I want to set the zero position for Z_axis based on thikness of the work area (usualy I use the milling head to make the work area or the bed flat)

### Additional context

In **M206_M428.cpp** I changed the function **void GcodeSuite::M428()** to make it as following, and I think it's working as I need, maybe this feature can be added to the original code as well if others need it.

```
void GcodeSuite::M428() {
  if (homing_needed_error()) return;

  xyz_float_t diff;
  LOOP_NUM_AXES(i) {
    if (parser.seen_test(AXIS_CHAR(i))){
      diff[i] = base_home_pos((AxisEnum)i) - current_position[i];
      if (!WITHIN(diff[i], -20, 20) && home_dir((AxisEnum)i) > 0)
        diff[i] = -current_position[i];
      if (!WITHIN(diff[i], -20, 20)) {
        SERIAL_ERROR_MSG(STR_ERR_M428_TOO_FAR);
        LCD_ALERTMESSAGE(MSG_ERR_M428_TOO_FAR);
        ERR_BUZZ();
        return;
      }
    }
    else {diff[i] = 0;}
    
  }

  LOOP_NUM_AXES(i) set_home_offset((AxisEnum)i, diff[i]);
  report_current_position();
  LCD_MESSAGE(MSG_HOME_OFFSETS_APPLIED);
  OKAY_BUZZ();
}
```

thisiskeithb, 2024-04-24:
> Maybe this feature can be added to the original code as well if others need it.

It would be useful if you could [submit a Pull Request](https://marlinfw.org/docs/development/getting_started_pull_requests.html) so it could be reviewed.

Elias-Ta, 2024-04-25:
Alright, I made a pull request [#27015](https://github.com/MarlinFirmware/Marlin/pull/27015)