MarlinFirmware/Marlin #27033: [FR] (feature summary) "HotBed" Idle Timeout

danihmorais, 2024-04-29:
### Describe the feature you want

Sorry bad English. I think that is cool to implement a function like HOTEND_IDLE_TIMEOUT, but for Hot Bed. Actually, this only verify the temperature hotend. It could be better, (for security too), if have a option for Hot Bed, avoid that a hot bed stay warming up forever. Suppose, for example, that I manually set 110º ONLY for the Hot BED, and not for hot end, and forget that I did that, it will stay there heating up, since the function HOTEND_IDLE_TIMEOUT will not be activated. 

If this already exists, could you show me where and how? Thank you.

classicrocker883, 2024-05-05:
Bed should get turned off at same time hotend idle timeout

**Marlin\src\feature\hotend_idle.cpp**

```y
void HotendIdleProtection::check() {
  const millis_t ms = millis();                   // Shared millis

  check_hotends(ms);                              // Any hotends need protection?
  check_e_motion(ms);                             // Motion will protect them

  // Hot and not moving for too long...
  if (next_protect_ms && ELAPSED(ms, next_protect_ms))
    timed_out();
}

// Lower (but don't raise) hotend / bed temperatures
void HotendIdleProtection::timed_out() {
  next_protect_ms = 0;
  SERIAL_ECHOLNPGM("Hotend Idle Timeout");
  LCD_MESSAGE(MSG_HOTEND_IDLE_TIMEOUT);
  HOTEND_LOOP() {
    if (cfg.nozzle_target < thermalManager.degTargetHotend(e))
      thermalManager.setTargetHotend(cfg.nozzle_target, e);
  }
  #if HAS_HEATED_BED
    if (cfg.bed_target < thermalManager.degTargetBed())
      thermalManager.setTargetBed(cfg.bed_target);
  #endif
}
```

thisiskeithb, 2024-05-05:
> Bed should get turned off at same time hotend idle timeout

Yes, but it relies on `HOTEND_IDLE_TIMEOUT` being enabled and above the set point. There’s no “`BED_IDLE_TIMEOUT`”-like feature where there’s a timeout if *only* the bed is heated.

danihmorais, 2024-05-05:
> Yes, but it relies on `HOTEND_IDLE_TIMEOUT` being enabled and above the set point. There’s no “`BED_IDLE_TIMEOUT`”-like feature where there’s a timeout if _only_ the bed is heated.

Exactly.