MarlinFirmware/Marlin #27038: [BUG] G28 causes incorrect Z-axis 0-point when using KlackEnder

Yangff, 2024-05-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I did some modifications to my Ender3 Pro and then I ran into strange problems. When homing with the G28, the print head would move to the center and then correctly probe z0's position and then lift up a distance.
Next, it raises to Z=20 and move to the far right, however, at this point the printer will not correctly move the print head to Z=0 to stow the probe, instead it will try to stow the probe at Z=20 and stop there, as it failed.
Even more strangely, this Z=20 showing on the display is not really Z=20, and if I open the motion menu at this point and move the print head to what the printer thinks is Z=0, the print head is a relatively large distance from the build plate.

[image]

As you can see here, at least after the probe ends, it marks the current Z as -1.2 (which is the correct 0+offset).  The Z0 is still correct at this time. 
And then stow fails.  The latter is when I use g0 to move Z to 0, but the print head doesn't move to 0.

Eventually I found that if I commented out `if (TERN0(HOMING_Z_WITH_PROBE, axis == Z_AXIS && probe.stow())) return;` and manually stowed the probe using the M402 at the end of the homing, at least the stow would do it correctly and the Z0 is right. I haven't tested if there will be any other problems when actually printing, but at least the homing is correct.

### Bug Timeline

at least 2.1.2.1

### Steps to Reproduce

I think repro this issue might be difficult because I modded more things. Everything worked fine when I started out with the KlackEnder, then I replaced the extruder, the fan, and then it started not working. Even with the same 2.1.2.1 firmware. 

I'm not quite sure what exactly is besides the problem, or if it's a combination of them that are causing the problem, I'd like to get some direction on debugging the problem, not that I can tell much from the current log. 

### Version of Marlin Firmware

2.1.2.2

### Printer model

Creality Ender-3 Pro

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/15176288/config.zip)

Yangff, 2024-05-01:
> Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

 I've actually never successfully booted with code under the bugfix branch. There is no output from the display and no output from the serial port.
I'm guessing it's a configuration item, because if I turn on `MARLIN_DEV_MODE` in the release branch it does the same thing.

ellensp, 2024-05-01:
Entry point 0x80090f1   
Its an elf, its relocatable to correct address.
This is fine

Much more likely you haven't checked what chip you have
The build environment you choose is stm32f103re or rc, the rc has  alot less flash and less ram.
building for a rc on a re chip works fine, but building for a re chip on a rc will not boot.

Yangff, 2024-05-01:
> Entry point 0x80090f1 Its an elf, its relocatable to correct address. This is fine
> 
> Much more likely you haven't checked what chip you have The build environment you choose is stm32f103re or rc, the rc has a lot less flash and less ram. building for a rc on a re chip works fine, but building for a re on a rc chip will not boot.
> 
> And the gd32 chip is just an incompatible nightmare...
> 
> So Identify your chip.

I‘m pretty sure it's a RC, the env I'm using is STM32F103Rx_creality. The bin is 190kb and the largest binary I have used with 2.1.2.2 is ~200kb so I don't think either is a problem.

The elf is not relocatable, as you can see the file type is EXEC..

ellensp, 2024-05-01:
you cannot build with STM32F103Rx_creality on real marlin, its an internal name.
If it does not start with  env: at the front you cannot use it as a build environment

[ELF](http://en.wikipedia.org/wiki/Executable_and_Linkable_Format) files are Executable Linkable Format which consists of a symbol look-ups and relocatable table, that is, it can be loaded at any memory address by the kernel and automatically, all symbols used, are adjusted to the offset from that memory address where it was loaded into. Usually ELF files have a number of sections, such as 'data', 'text', 'bss', to name but a few...it is within those sections where the run-time can calculate where to adjust the symbol's memory references dynamically at run-time.

Yangff, 2024-05-01:
I printed some logs..

```
DEBUG: homeaxis Z
>>> homeaxis(Z)
  current_position= X102.00 Y91.00 Z5.00 : Probe::set_deployed
deploy: 1
  current_position= X102.00 Y91.00 Z5.00 : Debug: Before z raise
Probe::do_z_raise(10.00)
echo:busy: processing
  current_position= X102.00 Y91.00 Z10.00 : Debug: After z raise
  current_position= X102.00 Y91.00 Z10.00 : Debug: old_xy
Home Fast: -375.00mm
Debug: Axis set to zero
echo:busy: processing
echo:busy: processing
Move Away: 5.00mm
Debug: Axis set to zero
echo:busy: processing
Re-bump: -10.00mm
Debug: Axis set to zero
echo:busy: processing
Re-bump done
Debug: Setting...
Debug: Axis Z is at home
  current_position= X102.00 Y91.00 Z-1.20 : sync_plan_position
  current_position= X102.00 Y91.00 Z-1.20 : > AFTER set_axis_is_at_home
  current_position= X102.00 Y91.00 Z-1.20 : Probe::set_deployed
deploy: 0
  current_position= X102.00 Y91.00 Z-1.20 : Debug: Before z raise
Probe::do_z_raise(10.00)
echo:busy: processing
  current_position= X102.00 Y91.00 Z10.00 : Debug: After z raise
  current_position= X102.00 Y91.00 Z10.00 : Debug: old_xy
  current_position= X102.00 Y91.00 Z10.00 : Debug: Before probe_specific_action
  current_position= X102.00 Y91.00 Z10.00 : Debug: Before stow step 1
echo:busy: processing
echo:busy: processing
  current_position= X245.00 Y114.00 Z20.00 : Debug: After stow step 1
  current_position= X245.00 Y114.00 Z20.00 : Debug: Before stow step 2
  current_position= X245.00 Y114.00 Z0.00 : Debug: After stow step 2
  current_position= X245.00 Y114.00 Z0.00 : Debug: Before stow step 3
  current_position= X220.00 Y114.00 Z0.00 : Debug: After stow step 3
  current_position= X220.00 Y114.00 Z0.00 : Debug: Before stow step 4
echo:busy: processing
echo:busy: processing
  current_position= X220.00 Y114.00 Z20.00 : Debug: After stow step 4
  current_position= X220.00 Y114.00 Z20.00 : Debug: After probe_specific_action
Error:Z-Probe failed
Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
  current_position= X220.00 Y114.00 Z20.00 : sync_plan_position
X:220.00 Y:114.00 Z:20.00 E:0.00 Count X:17600 Y:9120 Z:8000
ok
```

For some reason those
```
  current_position= X245.00 Y114.00 Z0.00 : Debug: After stow step 2
  current_position= X245.00 Y114.00 Z0.00 : Debug: Before stow step 3
```
are not executed on z-axis, but the printer thought they are done.
Any idea why this could happen? something to do with the planner?

closed 2024-08-14 by github-actions[bot]