MarlinFirmware/Marlin #27042: Fanuc m84 diş çemke kodu ekleyin

hsyngndz85, 2024-05-02:
### Is your feature request related to a problem? Please describe.

Fanuc makşnalarla uyum için m84 diş çekme gcode

### Are you looking for hardware support?

Kendi kartım

### Describe the feature you want

M84 komutu tornalarda diş çekmeye yarar

InsanityAutomation, 2024-05-02:
Do you mean G84?

https://gcodetutor.com/fanuc-training-course/g84-tapping-cycle.html

hsyngndz85, 2024-05-03:
M84 değil 
G84 fanuc tapping cycle

hsyngndz85, 2024-05-03:
Aslında işin özü şu G ve M komutlarının  linux cnc ve Fanuc için çakışan kodları. Ki aynı işi yapan komutlar var örneğin m226 pin control komutu linuxcnc de m66 komutu ikiside aynı işi yapıyor uyumluluk için bu iki sıtandartları marlinde birleştirip bunlarda olmayan 3d basım kodları bunların dışına itilip hem 3d sektörü hem linux cnc uyumlulu hem fanuc uyumluluğu sağlayan daha ileri bir marlin yapılabılır.şimdiye kadsr yaptığım araştırmalarda olmayan linux codları eklense linuxcnc nin üstüne çıkacak marlin neden linuxcnc de sd veya usbden dosya okuma komutu yok ve pc bağımlılığı var

hsyngndz85, 2024-05-03:
Bu G84 komutu ek bi özellik istiyor spindle motorunun da step motor gibi adımlı kontrol edilmesi gerekiyor yani spindle motoruda bir döner eksen gibi açısal sürülebilir olmalı tıpkı E0 motoru gibi

thisiskeithb, 2024-05-03:
@hsyngndz85: It would be really helpful if you could translate your Feature Requests & comments to English since that is the primary language used for this project.