MarlinFirmware/Marlin #27043: Hal esp32 de pin debug.h dosyası ekleyin

hsyngndz85, 2024-05-02:
### Is your feature request related to a problem? Please describe.

Pin debug ayarı açıldığında esp32 de hata veriyor

### Are you looking for hardware support?

Pin debug özelliği esp 32 de yok

### Describe the feature you want

Pin debug için gerekli h dosyası esp32 de yok

### Additional context

Eksik dosya

thisiskeithb, 2024-05-03:
@hsyngndz85: It would be really helpful if you could translate your Feature Requests & comments to English since that is the primary language used for this project.

hsyngndz85, 2024-05-03:
> @hsyngndz85: Bu projede kullanılan ana dil İngilizce olduğundan, Özellik İsteklerinizi ve yorumlarınızı İngilizceye çevirebilirseniz gerçekten yararlı olacaktır.

Ok

ellensp, 2026-10-30:
This was added to bugfix in https://github.com/MarlinFirmware/Marlin/pull/28550

closed 2026-10-30 by ellensp