MarlinFirmware/Marlin #27044: Esp32 PCF8575 kütüphane ekleyin

hsyngndz85, 2024-05-02:
### Is your feature request related to a problem? Please describe.

Fazladan rgiriş çıkış için pcf8575 desdeği ve m gcodlarıyla okuma yazma

### Are you looking for hardware support?

Kendi kartım

### Describe the feature you want

M coduyla fazladan giriş çıkış desdeği pcf8575

### Additional context

Pcf8575 için kutuphane mevcut

hsyngndz85, 2024-05-03:
Bu kitaplık hazır olması gereken komut seti şu 
Pin kontrol için M64 P0.4  pin derhal aç 
Pin contol için M65 P0.4 pin derhaö kapat
Pin durum bekle M66 P0.4
Komut aeti bu şekil linuxcnc uyumlu olur ki marlin uluslar arası bir gcode dosya uyumluluğu sağlar yani linuxcnc için yazılmış bi gcode sorunsuz çalışır

hsyngndz85, 2024-05-03:
https://linuxcnc.org/docs/devel/html/gcode/m-code.html
Bu sıtandartlara uyumlu bi marlin düzenlemesi yapılmalı ki 3d gcode seti artık kendine ait bir standart olmalı çakışmalar var buda marlini cnc torna gibi kullanmak istendiğinde sorun çıkartıyor.3d gcodeleri daha önce belirlenmiş linuxcnc ve fanuc standartının dışına çıkartılmalı bu gerçekleşirse marlin bu iki nc sisteminin üstünde daha ii bi hale getirir

hsyngndz85, 2024-05-03:
Bu özelliğin sağlıklı çalışır  şekilde eklenmesi esp32 gibi harika işlemcinin pin kısıtlaması olayını oratadan tamamen kaldıracak aşırı esnek hale getirecek .bu çip 16 bitlik bir çip ve 8 adete kadar adreslenebiliyor bu 8x16 dan toplam hem giriş hem çıkış olarak kullanılacak 128 pin demek yeterince ii bi olay.

thisiskeithb, 2024-05-03:
@hsyngndz85: It would be really helpful if you could translate your comments to English since that is the primary language used for this project.