MarlinFirmware/Marlin #27048: [BUG] Setting EXTRUDERS 0 fails to compile

opmtk, 2024-05-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I want to configure Marlin Fw without Extruder. Following [Issue #10212]( https://github.com/MarlinFirmware/Marlin/issues/10212) 'Zero extruders has been possible for a long while now.' which isn't true anymore.

Related issues/bugs are
- [Closed #9074](Bugfix 1.1.x Extruders = 0 make all Stepper available for X,Y,Z)
- [Closed #10217]([2.0.x] CONCEPT - Allow Extruders = 0 and use E0 socket for dual X, Y , Z stepper)

### Expected behavior

It's expected to compile.

### Actual behavior

Failed to compile with:
```
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:55,
                 from Marlin\src\HAL\STM32\HAL.cpp:26:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:117:26: error: static assertion failed: AXIS_RELATIVE_MODES must contain X Y Z elements.
  117 | static_assert(COUNT(arm) == LOGICAL_AXES, "AXIS_RELATIVE_MODES must contain " _LOGICAL_AXES_STR "elements.");
```

### Steps to Reproduce

1.  Change #define EXTRUDERS 1 to #define EXTRUDERS 0 in configuration.h
2.  Build

### Version of Marlin Firmware

tag 2.1.2.2 and bugfix-2.1.x branch

### Printer model

none

### Electronics

BTT Octopus v1.1

### LCD/Controller

none

### Other add-ons

none

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[MarlinConf.zip](https://github.com/MarlinFirmware/Marlin/files/15187766/MarlinConf.zip)

ellensp, 2024-05-02:
This is a user error

When you remove the extruder you MUST remove all extruder elements from the Configuration files

This example error in particular
The error "AXIS_RELATIVE_MODES must contain X Y Z elements"

But your AXIS_RELATIVE_MODES  is set to #define AXIS_RELATIVE_MODES { false, false, false, false }
Ie X,Y,Z and E elements

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

ellensp, 2024-05-02:
You have to remove the e parameter from all of these

DEFAULT_AXIS_STEPS_PER_UNIT
DEFAULT_MAX_FEEDRATE
MAX_FEEDRATE_EDIT_VALUES
DEFAULT_MAX_ACCELERATION
MAX_ACCEL_EDIT_VALUES
MAX_JERK_EDIT_VALUES
AXIS_RELATIVE_MODES
and probabbly some more

ellensp, 2024-05-02:
These are the changes you should made for your config to remove the extruder

```DIFF
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index 72ec272b4d..414d2f7883 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -68,7 +68,7 @@
 
 // Choose the name from boards.h that matches your setup
 #ifndef MOTHERBOARD
-  #define MOTHERBOARD BOARD_RAMPS_14_EFB
+  #define MOTHERBOARD BOARD_BTT_OCTOPUS_V1_1
 #endif
 
 /**
@@ -79,7 +79,7 @@
  *
  * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
  */
-#define SERIAL_PORT 0
+#define SERIAL_PORT -1
 
 /**
  * Serial Port Baud Rate
@@ -139,9 +139,9 @@
  *          TMC5160, TMC5160_STANDALONE
  * :['A4988', 'A5984', 'DRV8825', 'LV8729', 'TB6560', 'TB6600', 'TMC2100', 'TMC2130', 'TMC2130_STANDALONE', 'TMC2160', 'TMC2160_STANDALONE', 'TMC2208', 'TMC2208_STANDALONE', 'TMC2209', 'TMC2209_STANDALONE', 'TMC2660', 'TMC2660_STANDALONE', 'TMC5130', 'TMC5130_STANDALONE', 'TMC5160', 'TMC5160_STANDALONE']
  */
-#define X_DRIVER_TYPE  A4988
-#define Y_DRIVER_TYPE  A4988
-#define Z_DRIVER_TYPE  A4988
+#define X_DRIVER_TYPE  TMC2209
+#define Y_DRIVER_TYPE  TMC2209
+#define Z_DRIVER_TYPE  TMC2209
 //#define X2_DRIVER_TYPE A4988
 //#define Y2_DRIVER_TYPE A4988
 //#define Z2_DRIVER_TYPE A4988
@@ -153,7 +153,7 @@
 //#define U_DRIVER_TYPE  A4988
 //#define V_DRIVER_TYPE  A4988
 //#define W_DRIVER_TYPE  A4988
-#define E0_DRIVER_TYPE A4988
+//#define E0_DRIVER_TYPE A4988
 //#define E1_DRIVER_TYPE A4988
 //#define E2_DRIVER_TYPE A4988
 //#define E3_DRIVER_TYPE A4988
@@ -208,7 +208,7 @@
 
 // This defines the number of extruders
 // :[0, 1, 2, 3, 4, 5, 6, 7, 8]
-#define EXTRUDERS 1
+#define EXTRUDERS 0
 
 // Generally expected filament diameter (1.75, 2.85, 3.0, ...). Used for Volumetric, Filament Width Sensor, etc.
 #define DEFAULT_NOMINAL_FILAMENT_DIA 1.75
@@ -553,7 +553,7 @@
  *   998 : Dummy Table that ALWAYS reads 25°C or the temperature defined below.
  *   999 : Dummy Table that ALWAYS reads 100°C or the temperature defined below.
  */
-#define TEMP_SENSOR_0 1
+#define TEMP_SENSOR_0 0
 #define TEMP_SENSOR_1 0
 #define TEMP_SENSOR_2 0
 #define TEMP_SENSOR_3 0
@@ -1236,7 +1236,7 @@
  * Override with M92 (when enabled below)
  *                                      X, Y, Z [, I [, J [, K...]]], E0 [, E1[, E2...]]
  */
-#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 500 }
[... 87 lines cut ...]
     //#define Z3_STALL_SENSITIVITY Z_STALL_SENSITIVITY
     //#define Z4_STALL_SENSITIVITY Z_STALL_SENSITIVITY
-    //#define I_STALL_SENSITIVITY  8
-    //#define J_STALL_SENSITIVITY  8
-    //#define K_STALL_SENSITIVITY  8
-    //#define U_STALL_SENSITIVITY  8
-    //#define V_STALL_SENSITIVITY  8
+    #define I_STALL_SENSITIVITY  120
+    #define J_STALL_SENSITIVITY  120
+    #define K_STALL_SENSITIVITY  120
+    #define U_STALL_SENSITIVITY  120
+    #define V_STALL_SENSITIVITY  120
     //#define W_STALL_SENSITIVITY  8
     //#define SPI_ENDSTOPS              // TMC2130/TMC5160 only
     //#define IMPROVE_HOMING_RELIABILITY
@@ -3450,7 +3450,7 @@
    * Enable M122 debugging command for TMC stepper drivers.
    * M122 S0/1 will enable continuous reporting.
    */
-  //#define TMC_DEBUG
+  #define TMC_DEBUG
 
   /**
    * You can set your own advanced settings by filling in predefined functions.
```