MarlinFirmware/Marlin #27049: G code uyumlulu yok

hsyngndz85, 2024-05-03:
### Is your feature request related to a problem? Please describe.

Evet farklı örneğin linuxcnc ve fanucla g code çakışması var

### Are you looking for hardware support?

Hayır

### Describe the feature you want

Örneğin G10 komutu 3d dünyasında flement geri çekme komutu,
Ama linuxcnc ve fanucta takimlarla ilgili bir komut. Bunu önlemek için bi ayarla örneğin
Define 3d gcode standart eneble
Define cnc gcode standart eneble
Gibi bi ayar gerekli aksi halde marlini cnc dünyasında kullanamıyoruz.
Bu ayarlarla çakışan komutlar farklı makinalar için farklı yorumlanabilir.

### Additional context

Linuxcnc fanuc ve 3d g code standartları sorunu
https://linuxcnc.org/docs/html/gcode/g-code.html. Linuxcnc listesi
https://content.fanucworld.com/m-code-g-code-list/ fonuc cnc listesi
https://marlinfw.org/meta/gcode/ marlin listesi

thisiskeithb, 2024-05-03:
@hsyngndz85: It would be really helpful if you could translate your Feature Requests & comments to English since that is the primary language used for this project.

hsyngndz85, 2024-05-03:
Arık marlin sayesinde 3d gcode ve mcodeler bi standarta oturdu.
Bu kodların isimlerini değiştirmek imkansız ama config.h da bır ayarla gcode ve m codelerin hangi standarta göre yorumlanacağı eklenebilir

hsyngndz85, 2024-05-03:
M codelerde çakışma yok gibi sadece pin kontrolünde var
G codlarda baya çakışma var

thisiskeithb, 2024-05-03:
There may not be a way to make these compatible, but you can likely create a post-processing script to make them compatible or find someone who may have done this already.

hsyngndz85, 2024-05-03:
For example, when the m10 command is seen, it should do its job in 3D mode. If there is no equivalent for this command in Linux CNC mode, it should be stopped with m0. and write a message on the screen like (this command is not supported in this mode)

hsyngndz85, 2024-05-03:
I can solve this problem myself, but when M10 is seen, I cannot do what the M10 code should do in Linuxcnc mode. My software knowledge is limited.

hsyngndz85, 2024-05-03:
"#define Linuxcnc_mod" For example, in which file can I activate this? In this way, I can add a query in the gcode parser. When I receive a command that is not already known, Marlin says that it does not recognize this command, but it does not stop and moves to the next line.

hsyngndz85, 2024-05-03:
Let's say we put this setting in config.h, in which file can I assign this setting to the variable and query it in the g code parser?

hsyngndz85, 2024-05-03:
I tested it now, 3D commands are not recognized in CNC mode, but it does not stop and continues.How do we make the machine stop when an unrecognized command comes?

DerAndere1, 2024-05-03:
In general this feature would make sense. But regarding G10, there exist closed PRs that show how LinuxCNC-compatible handling of G10 L P can be added without running into conflicts.