MarlinFirmware/Marlin #27052: [BUG] MKS TinyBee Occasionally Fails to Execute G-code.

vfbank, 2024-05-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I would like to use this control board as functional hardware, but unfortunately, it's disappointing that many features cannot be used due to the limitations of the chipset.

Despite this, I've tried to use this control board and am currently using it reasonably well, except for one issue.

Yes, that one issue is a problem. It's about skipping G-code.

I am aware that there are issues when trying to access the ROM of the ESP32 chipset during layer shift, and so I have temporarily solved this problem by disabling the printer counter option.

But what really bothers me is that this chipset cannot properly process G-code.

The worst part is that sometimes, when auto-leveling (using a fixed-mount sensor) or auto-homing, it skips some movements. If I'm really unlucky, when executing auto-home, instead of moving to the center of the bed to find the Z-axis origin, it starts to descend from the point where the XY axis presses the limit sensor, as if doing manual leveling, and tries to destroy my printer.

There are no issues with the same firmware version on the STM32 chipset, and only the control boards with the ESP32 chipset show this phenomenon.

How can I solve this problem? Should I throw away this motherboard?

### Version of Marlin Firmware

2.1.x bugfix

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/15208349/Marlin.zip)

rondlh, 2024-05-05:
I don't think there is an problem with the hardware, there are many ESP32 based motherboards in the field that are running fine. Perhaps it's a communication issue or lose wire or something inconsistent like that.
You could try to setup a specific test for this, like making a gcode file that moves the nozzle in a square repeatedly and perhaps gives some feedback with M118, that might give you more insight in what's going wrong.

vfbank, 2024-05-06:
> I don't think there is an problem with the hardware, there are many ESP32 based motherboards in the field that are running fine. Perhaps it's a communication issue or lose wire or something [...]

Are you using hardware that uses the ESP32 chipset?

rondlh, 2024-05-09:
> Are you using hardware that uses the ESP32 chipset?

Yes, and in my experience it's reliable. ESP32 hardware is extremely common.

> What can only be seen as a hardware problem is not a problem at all on motherboards using Arduino Mega and STM32.
> The only thing that has changed is the chipset hardware.

Practically it's not as simple. Lot's of things can play a role, like lose contacts, wire length, the power supply and lots more.

> I've been using this motherboard for over a year and I'm still experiencing bugs but can't find a solution.

Did you do any specific tests to get more insight into where the problem is?

vfbank, 2024-05-09:
> Yes, and in my experience it's reliable. ESP32 hardware is extremely common.
>
>
> Practically it's not as simple. Lot's of things can play a role, like lose contacts, wire length, the power supply and [...]

I've tried everything over the past year, including grounding, power replacement, printing in different places, and shielding. And it's not just me; many other users are also suffering.

While the EPS32 chipset is used in various industries and is considered reliable, this is not the case with Marlin.

I've tried all the standard procedures mentioned not only by you but also those on Marlin and YouTube. However, I'm still facing issues where the 'busy' message and the OK message are not being displayed correctly.

Especially during auto-leveling, skipping the leveling pattern or crashing into the bed is truly horrendous.

I didn't know this motherboard was such a mess when I bought a large quantity of them, and I seriously regret it.

I've tried to make this motherboard work over the past year, but no one has been able to resolve the issue, and I've only been told to give up in the MKS Discord channels. Moreover, I still can't use features like baby stepping and input shaping, except for LA. This is already a recognized issue and has been marked in the issue tracker.

I don't understand why I have to suffer so much just because I used EPS32 when other motherboards don't have these problems.

If I send you this motherboard, could you fix it? I'm exhausted

noadgskill, 2024-05-09:
> I've tried everything over the past year, including grounding, power replacement, printing in different places, and shielding. And it's not just me; many other users are also suffering.
>
> While the [...]

I have the same problem. When the Tinybee configured printer uses auto leveling, it sometimes skips a leveling step or the nozzle scratches the bed because the Z offset is wrong. What's interesting is that I didn't have any problems with the Arduino board I was using initially. :)

I've removed this motherboard from the printer and am using it exclusively for CNC, but I would like to try using it again once the problem is solved.

rondlh, 2024-05-13:
> If I send you this motherboard, could you fix it? I'm exhausted

Where are you located?

closed 2024-09-02 by github-actions[bot]