MarlinFirmware/Marlin #27054: [BUG] (bug summary)

Daneild, 2024-05-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Trying to compile malin 2.1.2.2 for Anycubic kossell Lineal plus using a Octopus V1.1 and TMC2209 drivers, i will use a TFT35 display. I'm using default probe, it is a switch manually attached on the nozzle when calibrating.
I'm getting multiple errors of: #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."
I have found something close , an old post here but i'm not able to fix this. -> https://github.com/MarlinFirmware/Marlin/issues/16529
Thak you.

### Bug Timeline

from the  start

### Expected behavior

Expected to compile

### Actual behavior

Getting error:  #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."

### Steps to Reproduce

Download Marlin 2.1.2.2
Using config from: Configurations-release-2.1.2.2\config\examples\delta\Anycubic\Kossel Linear Plus
[2.1.2.2.zip](https://github.com/MarlinFirmware/Marlin/files/15214127/2.1.2.2.zip)

### Version of Marlin Firmware

2.1.2.2

### Printer model

Anycubic\Kossel Linear Plus

### Electronics

Octopus 1,1 , TMC2209

### LCD/Controller

TFT35

### Bed Leveling

ABL Linear grid

### Your Slicer

Slic3r

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

The final config i hope to have TFT35 with usb suport and TMC2209 drivers configured.
Maybe soft homing...but it is to hard for now at least.

ellensp, 2024-05-06:
add  #define NEEDS_Z_MINMAX to Your Configuration.h 

eg

```CPP
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BTT_OCTOPUS_V1_1
  #define NEEDS_Z_MINMAX
#endif

```

also you have in Configuration_adv.h "#define E0_AUTO_FAN_PIN 44 // turn off extruder fan when it's temperature is above 50°C"
Its pin 44. is not a valid fan pin. This will do unexpected things on an octopus board

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Daneild, 2024-05-07:
I have managed to compile, now there is something strage regarding endstops:

Using recomanded no jumpers on DIAG pins all endstops are triggered - with or no connection with switches

Putting DIAG first 2 jumpers ,( i think one corresponding to motor 0 and 1 ) the probe x_max, y_max, z_min are working OK 
but not the case with z_max:  that is always  TRIGGERED 

The DIAG jumpers are not working ok 

There any patch to this?