MarlinFirmware/Marlin #27056: [FR] M23 with long name

kleffa, 2024-05-06:
### Is your feature request related to a problem? Please describe.

M20L return long names, but in Marlin 2.1.2.2 with LONG_FILENAME_HOST_SUPPORT M23 only works with short name, that is not human friendly

### Are you looking for hardware support?

Serial comunnication, I have a skr 3 mini v2

### Describe the feature you want

M23 CEC2CD~1.GCO 
M23 CE3e08-quarto hexas 6-6h32m-133g.gcode
the same, also can be something like 
M35 CE3e08-quarto hexas 6-6h32m-133g.gcode
M23 CE3e08-quarto hexas 6-6h32m-133g.gcode L

### Additional context

some older versions worked with long names

ellensp, 2024-05-06:
Please test the [`bugfix-2.1.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

kleffa, 2024-05-06:
#define UTF_FILENAME_SUPPORT 
but not know if works in 2.1.2.2, but it was merged in bugfix

closed 2024-05-06 by kleffa