MarlinFirmware/Marlin #27062: [BUG] Unable to boot with any compiled commit after 2024-03-28 on the Ender 3 V2 Neo with an SKR3

vrooze, 2024-05-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Any commit since 2024-03-28 will not boot on my system.
They compile fine but when uploaded to the board it takes the firmware via the bootloader, renames it to `firmware.cur` and noting else happens.

I have bisected from the most recent commit all the way back to https://github.com/MarlinFirmware/Marlin/commit/17594298e194b14ee62d32d5574284b6b2c9eb5a
This is the last commit that works correctly.

I have raised a PR with the specific changes needed in order to get the board working at https://github.com/MarlinFirmware/Marlin/pull/27061

### Bug Timeline

2024-03-28

### Expected behavior

The compiled firmware to boot and work on the board.

### Actual behavior

Firmware compiles, but does not boot.

### Steps to Reproduce

Apply changes from the PR mentioned in the comments.
Set the board to `#define MOTHERBOARD BOARD_BTT_SKR_V3_0_EZ`
Compile the `STM32H723VG` version 
Upload the the board

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Creality Ender 3 V2 Neo

### Electronics

SKR 3 EZ

### LCD/Controller

DACAI 

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin-Configs.zip](https://github.com/MarlinFirmware/Marlin/files/15242850/Marlin-Configs.zip)

thisiskeithb, 2024-05-08:
> I have bisected from the most recent commit all the way back to [1759429](https://github.com/MarlinFirmware/Marlin/commit/17594298e194b14ee62d32d5574284b6b2c9eb5a)
> This is the last commit that works correctly.

That would make https://github.com/MarlinFirmware/Marlin/commit/466282f322ae4f42a8f3db5e3c9765c38445fdc8 / https://github.com/MarlinFirmware/Marlin/pull/26928 the breaking commit.

ellensp, 2024-05-09:
Please confirm, is this the board not talking or is it just the display not working so it seems dead. (ie do gcodes over usb to the controller work?)

dbombz, 2024-06-13:
I've encountered this issue as well. I determined that previous functional compilations were done with the environment  STM32H743Vx_btt in VScode. Now, that environment is no longer an option in ABM despite being set in platform.io. There are two options for environment now:  STM32H743VI_btt and STM32H723VG_btt. Compiling with ABM with either option results in a perpetually loading printer. 

If I attempt to compile with the check mark at the bottom of VSC, I get an error stating that 43Vx is incompatible, and I should use one of the other two (see pic).

[image]

Flashing with either binary, VI or VG, env results in a non-functional printer. I was able to reflash with my Vx binary and it works. The Vx I am using is 02010201.

ellensp, 2024-06-13:
@dbombz 

You don't randomly chose an environment.  You pick the one that matches the chip on your controller.
You have to look at the chip and read the number off the top, then use that build environment 

This is why there are now two options, this board now comes with two different chips a STM32H743VI or a STM32H723VG

So which chip is on your controller?

ellensp, 2024-06-13:
PROUI has known startup issue see 
https://github.com/MarlinFirmware/Marlin/issues/27141

ellensp, 2024-06-13:
Tested current bugfix using updated Configs from first post on a  STM32H743VI
It will not boot with PROUI enabled, even with patches mentioned in https://github.com/MarlinFirmware/Marlin/issues/27141 and https://github.com/MarlinFirmware/Marlin/pull/26917
Change it to Marlin UI and it boots fine.

dbombz, 2024-06-13:
I have the STM32H743VI. Recompiled, flashed, printed...all good.
Thank you for the enlightenment, and apologies for convoluting OP's issue.

 I recompiled with the VI environment selected. Everything appears all good...printing an 8 minute test benchy. Apologies for convoluting the issue.

vrooze, 2024-06-13:
I'm going to see if mines building for the wrong thing now!
I've got a 723Vx I hadn't noticed they were renamed without the x. Might explain why none of the UI options worked.

Will report back as soon as I can test!

vrooze, 2024-06-14:
Update.

Latest bugfix pulled. 
Still broken.
Swapped to MarlinUI and this time it did work!

I didn't check if serial comms worked with the first test as (as per) the initial issues when I raised this; No UI option worked and serial was also broken.

Looks like this issue has been semi resolved and is now just an issue with ProUI.

Will leave issue open so I can confirm a fix for as when the ProUI issue is solved, unless of course the powers that be would prefer it closed :)

thisiskeithb, 2024-06-14:
> Will leave issue open so I can confirm a fix for as when the ProUI issue is solved, unless of course the powers that be would prefer it closed :)

We already have an open bug report for ProUI, so I’m going to close this.

ellensp, 2024-06-16:
On this controller its really dead.. 

changed SERIAL_PORT to 1 and attaching a 3.3v serial to usb adapter  (-1 usb can lose data on startup and POSTMORTEM_DEBUGGING should also use this) 
I change the display over to LCD_SERIAL_PORT 3 so it didn't doesn't interfere with 1 the TFT port  (this is the wifi module UART) 

It doesn't say anything, not even a character!  

Other GUI's it boots up as expected logging to serial

ellensp, 2024-06-19:
Im reopening this.

If I reduce the config to basically just the motherboard and the proui and sdsupport and eeprom  and it boots again.

So im fairly sure this is something else

rzboril, 2024-06-24:
SKR3 723 is 550Mhz 
platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = STM32H723VG_btt

ellensp, 2024-06-24:
@rzboril REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER is not a serial device is a SPI device

rzboril, 2024-06-25:
> @rzboril REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER is not a serial device is a SPI device

My SKR 3 is conected both . LCD reprap and TFT.

ellensp, 2024-06-25:
@rzboril a Serial tft not the nothing like the e3v2 serial dwin display running proui of this issue.

Serail tfts talk standard gcode and are really just a dedicated control computer with display and touch interface. (and in this case 128x64 mono emulation) 

The e3v2 serial dwin display uses its own serial protocol, has it own user interface,  and doesn't have touch interface it only has an encoder,  and cannot emulate a 128x64. Ie compleatly different.

rzboril, 2024-08-06:
I just wanted to advise. It works like this for me without a problem.

closed 2024-12-29 by github-actions[bot]